stereolabs/zed-ros-wrapper #356: gravity vector not used in published frame?

eric-schleicher, 2018-12-27:
I'm integrating the zed mini into my capture system and for a variety of reasons the camera is oriented vertically.  When I use the ZED node, it would seem that the gravity vector is not used to establish the device orientation and my map data are not reflected correctly relative to ground truth.

[image]

It sort of seems that this should be sort of automatic and that i shouldn' have to publish and other transforms in the graph to get the data to be represented correctly.   Is there a setting to control this, or a workaround to correct the issue?

Myzhar, 2018-12-27:
Which version of the SDK are you using?
Have you updated to the latest `master` branch?

Myzhar, 2019-01-17:
Using the latest SDK [v2.7] and the latest version of the wrapper available in the `master` branch I cannot replicate the error.
Pointing the camera downward (not too much features to track on the ceiling) and moving around the gravity vector is correct and the path is correctly "horizontal".

closed 2019-01-17 by Myzhar