stereolabs/zed-ros-wrapper #360: How to improve the accuracy of /zed/odom in zed mini

EvanYehh, 2019-01-09:
Hi
We're trying to use /zed/odom to replace wheel odom in cartographer.
But we found out the /zed/odom will be draft after few times measurement.
We set up a zed mini and simply move 5m distance, move speed is slower than walk, straight forward to (5,0,0) then backward to (0,0,0)
At beginning the /zed/odom report is fine but it'll be worse after few times movement.
It'll be mush worse if there are people walking in front of zed mini.
When it draft, it might report x=9.xxx at the 5m goal.(it should be x= around 5) 
Is there any parameter to avoid this situation?

PS: use default config/launch file in zed_wrapper( set camera_model = 1 ZEDM)
roslaunch zed_wrapper zed.launch

Thanks,
Evan

Myzhar, 2019-01-09:
Can you record an SVO of the walking from 0m to 5m and share it with us together with the file `SNxxxx.conf` located in the folder `/usr/local/zed/settings`? 
In this way we can test it to try to understand what happens.

Meanwhile I suggest you to not replace the wheel odometry, but to fuse it with ZED odometry using a Kalman Filter (http://docs.ros.org/melodic/api/robot_localization/html/index.html)

EvanYehh, 2019-01-11:
Hi Myzhar
The SVO file and conf are in the google drive, please check it
https://drive.google.com/drive/folders/1FzoD5Dpu8mU45Of4MmA2o9xzax6N47ss?usp=sharing

Thanks,
Evan

Myzhar, 2019-01-11:
@EvanYehh a question: which version of the SDK are you using? Is it the latest v2.7 with the latest update in the master branch of the ROS wrapper?

EvanYehh, 2019-01-14:
Hi Myzhar
SDK is 2.6 and wrapper version as below
git branch -a -v
* master                        6c1810c Fixed inverted odometry base changing (#336)
  remotes/origin/HEAD           -> origin/master
  remotes/origin/dep_install_ci 25def17 add package update cmd CI
  remotes/origin/devel          79f9cac IMU minor changes
  remotes/origin/feat_obj_det   2a9c99b Enabled Object Detection publishing as objects.msg message
  remotes/origin/master         6c1810c Fixed inverted odometry base changing (#336)
  remotes/origin/nav_tutorial   cf1414c Merge branch 'master' into nav_tutorial

We setup a ZEDM on a robot, and we use keyboard_teleop to control robot.
(forward then turn around and forward, repeat few times)
So it might not be direct x-moving and exact 5m distance.
Will it better to use latest SDK and wrapper?
And is the vibration caused during robot moving a issue?

Thanks,
Evan

Myzhar, 2019-01-14:
Using the latest SDK and ROS wrapper is better for sure. There are a lot of improvements.

If you use pure VIO for odometry the vibrations can be a problem, this is why using a fused odometry is better

EvanYehh, 2019-01-15:
Hi Myzhar
We'll try to upgrade latest SDK and wrapper then verify again.

And how to add features to track? put some fixed items on wall or floor?
Will it be helpful to enable spatial memory and floor detection?

Thanks,
Evan

Myzhar, 2019-01-15:
Spatial Memory is not used for odometry. You can use it only if you fuse the ZED pose that is in `map` frame with other pose coming from other localization algorithms, for example AMCL.

The floor detection is useful to initialize the tracking with the real height from the floor.

Fixed items on the walls are helpful and they will improve the localization.

Myzhar, 2019-01-15:
@EvanYehh analyzing your SVO more in details we noticed that the optical calibration is not good, indeed walls and floor results curved:
[image]
[image]

You can fallow the guide here: https://support.stereolabs.com/hc/en-us/articles/360011828773-Can-I-recalibrate-my-ZED-camera-

Myzhar, 2019-04-01:
@EvanYehh can you update this issue and close it if your problem is solved?