stereolabs/zed-ros-wrapper #363: Publish rate of /zed/depth/depth_registered/compressedDepth

jsseng, 2019-01-16:
When I do a `rostopic hz /zed/depth/depth_registered`, I get an expected 30 Hz of messages.  When I do the same with the compressedDepth messages, I get .5 Hz of messages or no new messages.  I just was wondering if this is expected?

For the /zed/image_rect_color, I get about 30Hz and for the compressed version, I get 16Hz.  I am running on a Jetson TX2.  Thanks.

Myzhar, 2019-01-16:
The depth image is a floating point matrix, compressing it is an heavy work and the TX2 CPUs cannot handle it while performing all the other tasks.

If you want to transmit images over network I suggest you to use the topics throttle (http://wiki.ros.org/topic_tools/throttle) reducing the FPS and so the bandwidth.
You can use it with every kind of topic.

jsseng, 2019-01-16:
Got it.  Thank you for the quick reply and it will try it out.

closed 2019-01-16 by jsseng