stereolabs/zed-ros-wrapper #366: Problem reducing Point cloud data und accessing compressed depth image

Zhefei777, 2019-01-30:
Hello, 
We are doing a project with ZED, but I found the rate of topic point cloud is much slower than other topics. We are wondering, if it is possible to speed up the rate by reducing the accuracy? 
And I also want to ask how to get compressed depth image. 
Thanks in advance.

Zhefei777, 2019-01-31:
Thanks for your reply. I do measure the frequency with rostopic hz, and in my launch file the resolution is 3(RESOLUTION_VGA), mat_resize_factor is 1.0
But we got the frequency of point cloud about 3 hz while the frequency of image is 100hz.

Myzhar, 2019-01-31:
There is something weird, 3hz is really low.
What kind system are you using (CPU, RAM, GPU)? 
Are you using the latest version of the ROS wrapper and of the ZED SDK?

Zhefei777, 2019-02-01:
I used  Nvidia Jetson TX2. And now we decide not to use the point cloud  but just depth image. I want to ask how to get the compressed depth image. When I use rqt_image_view or rostopic echo to get the compressed depth image, there is nothing. 
Thank you  in advance.

Myzhar, 2019-02-01:
About the Jetson TX2: are you sure it has been set up to use all the computing power?
Read this: https://github.com/stereolabs/zed-ros-wrapper/issues/326

About the compressed depth I just tried and it works:
[image]

The problem is that the depth image is a floating point image and compressing it requires a lot of computational power... as you can see by the screenshot the frequency of the subscribed topic is very low even on a laptop. If you want to reduce the bandwidth required by the topic, I suggest you to use the topic_throttle tool:
https://github.com/stereolabs/zed-ros-wrapper/issues/363
https://github.com/stereolabs/zed-ros-wrapper/issues/292

Myzhar, 2019-02-01:
@Zhefei777 to reduce the computational power requested by compressedDepth you can play with Dynamic Reconfigure settings reducing the `png_level` value:
[image]

Zhefei777, 2019-02-05:
Thanks for your reply. I have an another question. The value I got from depth image is the straight distance between the object and the left lens of the camera or the projection of this distance onto the z axis?

Myzhar, 2019-02-05:
The depth image is in camera frame and the main camera is the left camera by default. 

The depth values are referred to the optical axis of the left camera: 
https://www.stereolabs.com/docs/positional-tracking/coordinate-frames/#selecting-a-coordinate-system

Zhefei777, 2019-02-05:
[image]
I am not sure. Could you tell me again, the depth value is just Zc or distance between M and Oc?  like this picture. 

`

Myzhar, 2019-02-05:
The image is not clear, but it can be more clear: 
the "depth" value is exactly the distance between the point measured and the plane of the left camera.

In camera frame (Z fwd, Y down, X right) it is the Z coordinate.

Zhefei777, 2019-02-05:
So all points at a plane , which is parallel with the plane of the left camera, have the same "depth" value, is it correct？

Zhefei777, 2019-02-05:
> The image is not clear, but it can be more clear:
> the "depth" value is exactly the distance between the point measured and the plane of the left camera.
> 
> In camera frame (Z fwd, Y down, X right) it is the Z coordinate.

All points in a plane , which is parallel with the plane of the left camera, have the same "depth" value, is it correct?

Myzhar, 2019-02-05:
Yes, that's correct

Zhefei777, 2019-02-05:
> Yes, that's correct

Thank you for your quick reply !!