stereolabs/zed-ros-wrapper #368: Unable to compile on Jetson TX2

varundevsukhil, 2019-02-03:
I have been trying to compile the ROS wrapper on JTX2 for a while now and I keep getting the same error. I have tried installing different SDK versions and different versions of the wrapper itself. Please help. Here is the error I get:

nvidia@tegra-ubuntu:~/zed_ws$ catkin_make
Base path: /home/nvidia/zed_ws
Source space: /home/nvidia/zed_ws/src
Build space: /home/nvidia/zed_ws/build
Devel space: /home/nvidia/zed_ws/devel
Install space: /home/nvidia/zed_ws/install
####
#### Running command: "cmake /home/nvidia/zed_ws/src -DCATKIN_DEVEL_PREFIX=/home/nvidia/zed_ws/devel -DCMAKE_INSTALL_PREFIX=/home/nvidia/zed_ws/install -G Unix Makefiles" in "/home/nvidia/zed_ws/build"
####
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/nvidia/zed_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/nvidia/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/nvidia/catkin_ws/devel;/opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/nvidia/zed_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 9 packages in topological order:
-- ~~  - zed_ar_track_alvar_example
-- ~~  - zed_depth_sub_tutorial
-- ~~  - zed_display_rviz
-- ~~  - zed_nodelet_example
-- ~~  - zed_ros (metapackage)
-- ~~  - zed_rtabmap_example
-- ~~  - zed_tracking_sub_tutorial
-- ~~  - zed_video_sub_tutorial
-- ~~  - zed_wrapper
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'zed_ar_track_alvar_example'
-- ==> add_subdirectory(zed-ros-wrapper/examples/zed_ar_track_alvar_example)
-- +++ processing catkin package: 'zed_depth_sub_tutorial'
-- ==> add_subdirectory(zed-ros-wrapper/tutorials/zed_depth_sub_tutorial)
-- +++ processing catkin package: 'zed_display_rviz'
-- ==> add_subdirectory(zed-ros-wrapper/zed_display_rviz)
-- +++ processing catkin package: 'zed_nodelet_example'
-- ==> add_subdirectory(zed-ros-wrapper/examples/zed_nodelet_example)
-- +++ processing catkin metapackage: 'zed_ros'
[... 115 lines cut ...]
/opt/ros/kinetic/include/ros/console.h:568:43: note: in expansion of macro ‘ROS_LOG_STREAM_COND’
 #define ROS_LOG_STREAM(level, name, args) ROS_LOG_STREAM_COND(true, level, name, args)
                                           ^
/opt/ros/kinetic/include/rosconsole/macros_generated.h:59:32: note: in expansion of macro ‘ROS_LOG_STREAM’
 #define ROS_DEBUG_STREAM(args) ROS_LOG_STREAM(::ros::console::levels::Debug, ROSCONSOLE_DEFAULT_NAME, args)
                                ^
/home/nvidia/zed_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:2259:17: note: in expansion of macro ‘ROS_DEBUG_STREAM’
                 ROS_DEBUG_STREAM(sl::toString(compression).c_str() << "not available. Trying " << sl::toString(
                 ^
/home/nvidia/zed_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:2261:31: error: ‘SVO_COMPRESSION_MODE_AVCHD’ is not a member of ‘sl’
                 compression = sl::SVO_COMPRESSION_MODE_AVCHD;
                               ^
/home/nvidia/zed_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:2264:28: error: ‘ERROR_CODE_SVO_UNSUPPORTED_COMPRESSION’ is not a member of ‘sl’
                 if (err == sl::ERROR_CODE_SVO_UNSUPPORTED_COMPRESSION) {
                            ^
/home/nvidia/zed_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:2277:20: error: ‘ERROR_CODE_SVO_UNSUPPORTED_COMPRESSION’ is not a member of ‘sl’
         if (err == sl::ERROR_CODE_SVO_UNSUPPORTED_COMPRESSION) {
                    ^
zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/build.make:86: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o' failed
make[2]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o] Error 1
CMakeFiles/Makefile2:2121: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j6 -l6" failed

Myzhar, 2019-02-03:
Have you installed the latest SDK v2.7.1 available on Stereolabs website?

The error is on `SVO_COMPRESSION_MODE_HEVC` that is available only since v2.7.

Let me know, meanwhile we will check again on the TX2

rechie901, 2019-02-08:
Hi, I am facing the same issue with SDK 2.5.1 and the latest ROS wrapper.

Myzhar, 2019-02-08:
This is a known bug. It will be fixed with the next release. Meanwhile you can change this line:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L2253

this way:
```#if ((ZED_SDK_MAJOR_VERSION>2) || (ZED_SDK_MAJOR_VERSION==2 && ZED_SDK_MINOR_VERSION>=7))```

and recompile

varundevsukhil, 2019-02-18:
Thanks @Myzhar 

That solved the problem

closed 2019-02-18 by varundevsukhil

mhmdkanj, 2019-02-21:
I tried what @Myzhar suggested but the error persisted. I am using ZED SDK 2.4
Does anyone know what the problem is?

Myzhar, 2019-02-21:
@mhmdkanj are you getting the same exact error? The SDK v2.4 is pretty old, maybe the error is another.
Please open a new issue and paste the compiling error