stereolabs/zed-ros-wrapper #370: Get the 3D point cloud values for pixel (i, j)

Wotipati, 2019-02-11:
Hello Everyone,

Could anyone tell me how to get the 3D point cloud values for pixel (i, j)?

Should I use the ZED API (please see the following link) rather than zed-ros-wrapper?
(https://www.stereolabs.com/docs/depth-sensing/using-depth/#getting-point-cloud-data)

Thanks,

Seita

Myzhar, 2019-02-11:
There are many ways to get the values of a pixel of a point cloud, depends on how you want to get it: direct access to the ZED SDK inside the ZED ROS wrapper node, subscribe to a point cloud in an external ROS node, you want only the XYZ coordinate of the point, you need also the RGB value, ...

Can you specify better your requirements such that we can help you?

Wotipati, 2019-02-11:
Thank you for your prompt response.

I want the XYZ value (don't need the RGB value) for a pixel (i, j) of RGB Image recorded by the ZED camera.
(Don't have to subscribe to a point cloud in an external ROS node)

Thanks,

Myzhar, 2019-02-11:
So this is not a ROS question, you are in the wrong place.

You can find the reply to your question in this GIT repository:
https://github.com/stereolabs/zed-examples/tree/master/depth%20sensing

Wotipati, 2019-02-11:
My apologies for the comment which might have been misleading. I should have explained it better so let me rephrase it.

I want to write a simple node that subscribes to messages of tye sensor_msgs/PointCloud2 to retrieve the point cloud published by the ZED node and to get the XYZ value for a pixel (i,j).

```
#include <ros/ros.h>
#include <sensor_msgs/PointCloud2.h>

float x, y, z;
int i, j;

/* -------------------- callbacks -------------------- */
void pclCallback(const sensor_msgs::PointCloud2::ConstPtr& msg)
{
	/* Get the XYZ value for a pixel (i,j) */
	x = .......
	y = .......
	z = .......

}

int main(int argc, char** argv) {
    
    ros::init(argc, argv, "zed_video_subscriber");

    ros::NodeHandle n;
    ros::Subscriber subPCL = n.subscribe("/zed/point_cloud/cloud_registered", 10, pclCallback);
	

    /* Some processing using x, y, z, i, and j */
    ......

    ros::spin();

    return 0;
}
```

Best,

Myzhar, 2019-02-11:
Ok, so this can help you:
https://answers.ros.org/question/191265/pointcloud2-access-data/

There is a link where you can find a `getXYZ` function that is what you are searchin for: https://pastebin.com/i71RQcU2

Let me know if that helps you

Wotipati, 2019-02-11:
It's Perfect!!! :)

Thank you so much!!!

Myzhar, 2019-02-11:
You are welcome

closed 2019-02-11 by Myzhar