stereolabs/zed-ros-wrapper #371: Error when trying to record svo with the rosservice

AlphaGolf99, 2019-02-15:
Hello, i have installed the latest version of ZED SDK 2.7.1 and I want to record a SVO while tracking the position of the camera using the ROS wrapper.
The start_svo_recording service launched correctly and is waiting to record but when I subscribe to the /zed/odom topic instead of recording I get the following error : "Error saving frame to SVO"
Does someone know how to fix this ?

Myzhar, 2019-02-18:
Hi @AlphaGolf99, 
the current version of the wrapper requires that you subscribe at least a video or a depth topic to grab the frames to be recorded. 
We will change the behavior to automatically grab the frames when the start SVO recording service is called.

AlphaGolf99, 2019-02-18:
Hello, thank you for the fix I tried it but I always get the same error.

Also before the fix, as you said I subscribed to /zed/depth/depth_registered or /zed/rgb/image_raw_color, while the recording service was waiting but still get the error.

Myzhar, 2019-02-18:
The fix has not yet been merged in the master branch. You must checkout the `fix_371` branch to test it before the merging

AlphaGolf99, 2019-02-18:
Yes I did the modification on my side and rebuilt it. When I call start_svo_recording instead of having the waiting status I get the error status in the topic /diagnostics

Myzhar, 2019-02-18:
I'm going to check it. Can you tell me the exact error you are getting?

AlphaGolf99, 2019-02-18:
Here is a screenshot from my desktop while I run the svo recording service.
The error I get is "Error saving frame to svo"

[image]

Myzhar, 2019-02-18:
Ok, I found your problem:
`~/Desktop`
You must use full path. The ZED SDK cannot resolve the `~`

AlphaGolf99, 2019-02-18:
I still get the error messages but now the svo file is correctly populated, I can use the playback example to review it or apply the positional tracking on the svo and it works.

Myzhar, 2019-02-18:
I tried to reproduce the error, but I cannot.

[image]

AlphaGolf99, 2019-02-18:
I am working on ROS kinetic and not melodic, maybe this is the problem ?

Myzhar, 2019-02-18:
No, Kinetic is fully compatible. Melodic still has some issues and not all the packages are available yet

AlphaGolf99, 2019-02-18:
I will try to reinstall the SDK and the wrapper, maybe there was problem. I will let you know.
Thank you for your work and for your time

AlphaGolf99, 2019-02-18:
I reinstalled the SDK 2.7.1 and the ros wrapper, but nothing changed, I still have the error messages but at least the svo is recording correctly

closed 2019-02-20 by Myzhar

killkelleyr, 2019-04-10:
I dont know if this would fall under this issue however when I run the start_svo_recording it successfully creates the SVO file but then when I go to open the SVO in ZEDfu it crashes telling me "Invalid Mesh, mesh does not contain enough vertices"

Myzhar, 2019-04-10:
@killkelleyr have you tried to open it with "ZED Depth viewer" to see if the depth information are correct?

killkelleyr, 2019-04-10:
@Myzhar I am not sure where to find "ZED Depth Viewer" would this be a ROS service?

Myzhar, 2019-04-10:
It's one of the ZED tools in /usr/local/zed/tools

killkelleyr, 2019-04-10:
When we try to open our SVO file in the "ZED Depth Viewer" it appears to show us a depth map. However, the video quality is extremely low. Do you think it could be something to do with the video quality?

Myzhar, 2019-04-10:
Are you using a TX2? How long is the recording?

killkelleyr, 2019-04-10:
Yes we are using the TX2, this recording was about 20 seconds long

killkelleyr, 2019-04-10:
@Myzhar after running another SVO recording with the ROS wrapper we noticed we got a bunch of errors

`[ INFO] [1554915264.707525204]: SVO recording STARTED: Test4.svo (HEVC (H265))
[ INFO] [1554915264.715419064]: *** Starting Positional Tracking ***
[ INFO] [1554915264.724000140]: Init Odometry with first valid pose data : ENABLED
[ INFO] [1554915264.726249175]: Pose Smoothing : DISABLED
[ INFO] [1554915264.726410451]: Spatial Memory : DISABLED
[ INFO] [1554915264.726485201]: Two D mode : DISABLED
[ INFO] [1554915264.726560623]: Floor Alignment : DISABLED
[ INFO] [1554915280.285480208]: Tracking ENABLED
ZED (Init) >> WARNING: FPS is too low to enable positional tracking and spatial mapping. Consider using PERFORMANCE parameters
[ERROR] [1554915280.332538707]: Error saving frame to SVO
[ INFO] [1554915280.337504797]: Odometry aligned to last tracking pose
ZED (Init) >> WARNING: FPS is too low to enable positional tracking and spatial mapping. Consider using PERFORMANCE parameters
ZED (Init) >> WARNING: FPS is too low to enable positional tracking and spatial mapping. Consider using PERFORMANCE parameters
[ERROR] [1554915281.349075320]: Error saving frame to SVO
[ERROR] [1554915282.350205484]: Error saving frame to SVO
[ERROR] [1554915283.380612204]: Error saving frame to SVO
[ERROR] [1554915284.391432895]: Error saving frame to SVO
[ERROR] [1554915285.416395205]: Error saving frame to SVO
[ERROR] [1554915286.419938793]: Error saving frame to SVO
[ERROR] [1554915287.422730592]: Error saving frame to SVO
[ERROR] [1554915288.453633887]: Error saving frame to SVO
[ERROR] [1554915289.485268911]: Error saving frame to SVO
[ERROR] [1554915290.518232609]: Error saving frame to SVO
[ERROR] [1554915291.549631899]: Error saving frame to SVO
[ERROR] [1554915292.555721743]: Error saving frame to SVO
[ERROR] [1554915293.557015254]: Error saving frame to SVO
[ERROR] [1554915294.584033918]: Error saving frame to SVO
[ERROR] [1554915295.587687122]: Error saving frame to SVO
[ERROR] [1554915296.616803788]: Error saving frame to SVO
[ERROR] [1554915297.620669855]: Error saving frame to SVO
[ERROR] [1554915298.650330960]: Error saving frame to SVO
[ERROR] [1554915299.680747314]: Error saving frame to SVO
[ INFO] [1554915300.052669286]: SVO recording STOPPED
`

Myzhar, 2019-04-10:
Verify a few things:
1. You are running the TX2 at maximum power (run the jetson_clocks.sh script)
2. You have enough ROM space
3. You are not requesting too high framerate (best using 15) or resolution (HD720)

To record a good SVO on TX2 the recording must last at least 30 seconds to allow the calculation of the better bitrate

killkelleyr, 2019-04-10:
@Myzhar  when we run the jetson_clocks.sh script it comes up with the error 
`bash: jetson_clocks.sh: command not found` 

are we missing a library perhaps?

Myzhar, 2019-04-10:
If you are on a Jetpack version minor than 4 the script is located in the home folder of the `nvidia` user
Try
`$ sudo ~/jetson_clocks.sh`

killkelleyr, 2019-04-10:
@Myzhar  We are currently running Jetpack 4.2
`#R32 (release), REVISION: 1.0`

killkelleyr, 2019-04-10:
@Myzhar we figured out that on Jetpack 4.2 jetson_clocks is an executable. It successfully ran and we are now running at max power.

However after we ensured we were running the ZED at HD720p at 15fps, we are still getting the error
`Error saving frame to SVO`

Myzhar, 2019-04-10:
You can ignore the warning if it appears only when you start the node (it's a sync issue solved with the next version of the wrapper).

About the SVO problem... have you set a relative path to save it?

killkelleyr, 2019-04-10:
@Myzhar I do not think so as we are only saving it into the .ros folder which is default. How would we set the relative file path?

zhangliyun9120, 2021-04-15:
@Myzhar I instlled the zed sdk 3.4, and all command are right, I still got the error:
**Error saving frame to SVO**
the zed sdk 3.4 node is different, can you check this? thank you.

[image]

Myzhar, 2021-04-15:
@zhangliyun9120 please open a new issue adding more info about you system