stereolabs/zed-ros-wrapper #374: NaN Values in Depth subscribing examples

qblark, 2019-02-19:
I know there are other issues that contain this same issue but none of them have addressed my concerns satisfactorily. I receive "Nan" while computing the average distance to camera at a range of points in the camera. Is there a way to force the zed-ros-wrapper to estimate the distance or remove the NaN values? The nan values are causing an inconsistency in my robot. Thank you in advance

Myzhar, 2019-02-19:
Not all the pixels of a depth map contains valid values (object occlusions, prospective, light flares, ...) and the not valid pixels have Nan values.
To get a valid mean distance you must filter the Nan values.

qblark, 2019-02-19:
Ok, thank you

closed 2019-03-04 by Myzhar