stereolabs/zed-ros-wrapper #377: Unable to compile ZED ROS Wrapper on Jetson TX2

mhmdkanj, 2019-02-21:
After following the steps to install the ZED ROS Wrapper on my Jetson TX2, the build of the wrapper gives an error when I enter catkin_make 
(Jetson flashed with Jetpack 3.3)
(CUDA 9.0)
(ZED SDK 2.4) [There were problems while installing the 2.7, which forced me to use the 2.4]
(Ubuntu 16.04)
It would have something to do with the UNSUPPORTED SVO bug 
```
nvidia@tegra-ubuntu:~/catkin_ws$ catkin_make
Base path: /home/nvidia/catkin_ws
Source space: /home/nvidia/catkin_ws/src
Build space: /home/nvidia/catkin_ws/build
Devel space: /home/nvidia/catkin_ws/devel
Install space: /home/nvidia/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/nvidia/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/nvidia/catkin_ws/build"
####
[  5%] Built target zed_video_sub
[ 10%] Built target zed_depth_sub
[ 15%] Built target zed_tracking_sub
[ 15%] Built target _zed_wrapper_generate_messages_check_deps_reset_tracking
[ 15%] Built target _zed_wrapper_generate_messages_check_deps_set_initial_pose
[ 15%] Built target _zed_wrapper_generate_messages_check_deps_start_svo_recording
[ 15%] Built target _zed_wrapper_generate_messages_check_deps_stop_svo_recording
[ 17%] Built target zed_wrapper_gencfg
[ 17%] Built target _zed_wrapper_generate_messages_check_deps_reset_odometry
[ 33%] Built target zed_wrapper_generate_messages_py
[ 46%] Built target zed_wrapper_generate_messages_lisp
[ 58%] Built target zed_wrapper_generate_messages_nodejs
[ 71%] Built target zed_wrapper_generate_messages_cpp
[ 87%] Built target zed_wrapper_generate_messages_eus
Scanning dependencies of target ZEDWrapper
[ 87%] Built target zed_wrapper_generate_messages
[ 89%] Building CXX object zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o
/home/nvidia/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘bool zed_wrapper::ZEDWrapperNodelet::on_start_svo_recording(zed_wrapper::start_svo_recording::Request&, zed_wrapper::start_svo_recording::Response&)’:
/home/nvidia/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:2331:20: error: ‘ERROR_CODE_SVO_UNSUPPORTED_COMPRESSION’ is not a member of ‘sl’
         if (err == sl::ERROR_CODE_SVO_UNSUPPORTED_COMPRESSION) {
                    ^
zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/build.make:86: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o' failed
make[2]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o] Error 1
CMakeFiles/Makefile2:2121: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```

Myzhar, 2019-02-21:
The SDK v2.4 is a very old version. 
Updating to the most recent v2.7.1 will bring a lot of improvements. 
What kind of problems are you getting installing the latest version?

Meanwhile we will work to fix v2.4 compatibility

mhmdkanj, 2019-02-21:
@Myzhar When I installed ZED SDK 2.7, it seemed to have done so. However, when I tried to double-click on a ZED tool (such as ZED Explorer), nothing would happen - not even an error would flash (I set the ZED Explorer to run as executable file in the properties), but nothing I tried solved this, except installing the older v2.4.

Do you suggest updating to v2.7 amidst my build of the catkin_make or would it disrupt the whole ROS kinetic/catkin environment setup?

Myzhar, 2019-02-22:
If you install the v2.7 and recompile you ROS workspace everything should work.

I change SDK version continuously for testing and there are no issues doing it

mhmdkanj, 2019-02-22:
@Myzhar Yes that worked perfectly thank you!

However, I am now experiencing the same problem on my Ubuntu 16.04 desktop, where upon installing any ZED SDK version, double-clicking on a ZED tool (e.g. ZED Explorer) doesn’t do anything. Would you have any idea why this could be? 

It is much appreciated!

Myzhar, 2019-02-22:
I'm checking it on the TX2 and let you know

Myzhar, 2019-02-22:
@mhmdkanj can you try to start ZED Explorer in a terminal, you can get the error stream and have more information about the problem:
`$ "/usr/local/zed/tools/ZED Explorer"`

mhmdkanj, 2019-02-22:
@Myzhar I tried that; however, it showed this error which I'm not knowing how to solve:
```
mohamad@mohamad-Inspiron-13-7359:/usr/local/zed/tools$ '/usr/local/zed/tools/ZED Explorer' 
/usr/local/zed/tools/ZED Explorer: error while loading shared libraries: libhidapi-libusb.so.0: cannot open shared object file: No such file or directory
```

Myzhar, 2019-02-22:
Ok, the error is self explaining. You miss the libusb library. Try to install it using apt:
`$ sudo apt install libusb-1.0-0`

mhmdkanj, 2019-02-22:
@Myzhar I already did that but it still shows the same error:
```
mohamad@mohamad-Inspiron-13-7359:/usr/local/zed/tools$ '/usr/local/zed/tools/ZED Explorer' 
/usr/local/zed/tools/ZED Explorer: error while loading shared libraries: libhidapi-libusb.so.0: cannot open shared object file: No such file or directory
mohamad@mohamad-Inspiron-13-7359:/usr/local/zed/tools$ sudo apt install libusb-1.0-0
[sudo] password for mohamad: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libusb-1.0-0 is already the newest version (2:1.0.20-1).
0 upgraded, 0 newly installed, 0 to remove and 64 not upgraded.
```

Myzhar, 2019-02-22:
Check my message, I changed the lib to be installed

mhmdkanj, 2019-02-22:
Sorry to say that I just tried that and still getting the same error:
```
mohamad@mohamad-Inspiron-13-7359:/usr/local/zed/tools$ '/usr/local/zed/tools/ZED Explorer' 
/usr/local/zed/tools/ZED Explorer: error while loading shared libraries: libhidapi-libusb.so.0: cannot open shared object file: No such file or directory
mohamad@mohamad-Inspiron-13-7359:/usr/local/zed/tools$ sudo apt install libusb-dev
[sudo] password for mohamad: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libusb-dev is already the newest version (2:0.1.12-28).
0 upgraded, 0 newly installed, 0 to remove and 64 not upgraded.
```

(I am rebooting after every trial just in case)

Myzhar, 2019-02-25:
@mhmdkanj all the dependencies should be installed with the SDK installation script. 
Have you refused to install dependencies during installation?
Try to install these libraries and let me know it they solve the problem:
`$ sudo apt install libhidapi-dev libhidapi-libusb0`

mhmdkanj, 2019-02-25:
@Myzhar Yes exactly that solved the problem! Much appreciated

Myzhar, 2019-02-25:
You are welcome

closed 2019-02-25 by Myzhar