stereolabs/zed-ros-wrapper #378: poor quality video when recording svo files using rosservice

NikoBach2, 2019-02-22:
Hi 
I have updated my zed_wrapper after your latest update.
I have tried to record an SVO file without any subscriber. The SVO file is recorded, but when i open the file in ZED Explorer it has a very poor resolution. It also says that the frame_rate is 3, even though i have set it to 30. The resolution is set to 1280x720
If i compare the total number of frames with the time i have recorded, i would get around 30 FPS. But if i read the frame-rate in ZED Explorer, or read it with the getSVONumberOfFrames() function, it writes out 3 FPS.
If i have a node subscribing to one of the topics, i have no problems.
I have attached an svo file were it fails.

[GrapWithoutSubcriber_camera1.svo.zip](https://github.com/stereolabs/zed-ros-wrapper/files/2893885/GrapWithoutSubcriber_camera1.svo.zip)

Myzhar, 2019-02-22:
@NikoBach2 thank you for your submission.
We will check the issue and fix it

Myzhar, 2019-02-22:
@NikoBach2 I tested your SVO and I get 3 fps too and a very poor quality.
Can you give me more information about the system you used to record it?

I just tried a recording using your configuration and I correctly get 30FPS @ HD720.

The only difference I noticed is about the compression algorithm. My system does not support H265 and automatically switched to H264.

axjensen, 2019-02-22:
@Myzhar . I am working together with @NikoBach2 . The recording was made on a Jetson TX1. As far as I recall, the Jetson TX1 also has limitations on compression algorithms. We can add more details as soon as possible

obraun-sl, 2019-02-22:
Hi,
Currently the H26x SVO compression uses the resolution and fps to calculate the bitrate and that can lead to some issues of quality when FPS is low. 
On TX1/2, the FPS is quite low at the beginning of the grab loop, therefore it could be better to wait for some grab calls before calling enableRecording so that the camera FPS is stabilized at the requested value (15fps or more) .

OB.

NikoBach2, 2019-02-25:
Okay, that explains it.  
Thank you very much for your response.

closed 2019-02-25 by NikoBach2