stereolabs/zed-ros-wrapper #381: mRecState.status always false

AlphaGolf99, 2019-02-26:
Hello,

Following the issue #371 where I had errors while saving svo files using the ros service even though the recording worked fine.

I found the problem, it is the mRecState.status variable that should be true after mzed.record() executed correctly but is always false. Thing is I believe the code of the record() function is hidden as I can't find it anywhere, so I can't check what is wrong.

I am using the SDK 2.7.1, just in case I reinstalled it last week but I still have the same problem. Also the record().status is always false even when I use the C++ code for recording.

Do you have any idea how to correct this or should I contact the stereolabs support ?

Myzhar, 2019-02-26:
Hi @AlphaGolf99 can you point me to the part of the code that generates this issue?

AlphaGolf99, 2019-02-26:
It is in Camera.hpp
`RecordingState record()` line 1932

Myzhar, 2019-02-26:
Line 1932 does not contain that code. Did you mean this one?
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L1779

AlphaGolf99, 2019-02-26:
Yes sorry I mistook the ZED SDK and ros wrapper

Myzhar, 2019-02-26:
I tried again, but I cannot replicate your problem. 
I start the node:
`$ roslaunch zed_wrapper zed.launch`
Then I start the recording:
`$ rosservice call /zed/start_svo_recording "svo_filename: 'test.svo'"`
and everything works as expected: no error message at all.

Can you post the exact procedure you are using (with the exact commands) ?

AlphaGolf99, 2019-02-26:
Sure
`$ roscore`
`$ roslaunch zed_wrapper zed.launch`
`$ rostopic echo /zed/odom`
`$ rosservice call /zed/start_svo_recording filename:='/media/nvidia/9016-4EF8/ZED_data/file.svo`

Myzhar, 2019-02-26:
So you are recording on an USB key. 
Is the problem still present if you record on the HD/ROM?

AlphaGolf99, 2019-02-26:
Yes but I have limited free space on the hard drive

Myzhar, 2019-02-26:
Are you getting the same error on the hard drive?

AlphaGolf99, 2019-02-26:
yes

Myzhar, 2019-02-26:
Have you updated to the latest version available in the `master` branch and then recompiled?
Just to be sure, because I cannot replicate it in any way

AlphaGolf99, 2019-02-26:
I will try it. I think I have a 7 days old version. But from what I investigated the problem comes from within the record() function in the SDK.

Myzhar, 2019-02-26:
After you have updated, if you continue to face the problem please post me the exact configuration of your system. I will forward it to the dev support to check it

AlphaGolf99, 2019-02-26:
I still have the same errors whether I record on the Hard Drive or on the SD card.
We use a jetson TX2 with tegra leopard 4.4.38, ubuntu 16.04, gcc version 4.8.5, 4 ARMv8 CPUs, aarch64.

AlphaGolf99, 2019-02-26:
I forgot ROS kinetic and CUDA 9.0

Myzhar, 2019-02-26:
I'm going to test it again on the Jetson TX2 and forward the problem to SDK developers.

AlphaGolf99, 2019-02-26:
ok thank you

closed 2019-07-10 by Myzhar