stereolabs/zed-ros-wrapper #384: Question on imu URDF as floating joint

rhklite, 2019-03-08:
I'm seeing that the imu_link_joint defined as a floating joint. From my understanding, I thought that floating joint woud mean the IMU could be moving with respect to to the zedm camera. However, this is should not be the case, as the IMU is rigidly fixed to the zedm camera. So does that mean this joint should be `type="fixed"`?

Would this understanding be correct?

```urdf
<!-- IMU -->    

    <joint name="imu_link_joint" type="floating">
        <parent link="zed_camera_center"/>
        <child link="imu_link"/>
        <origin xyz="0 0 0" rpy="0 0 0" />
    </joint>
    <link name="imu_link" />
```

Myzhar, 2019-03-08:
Hi @rhklite, what you are saying would be correct if the IMU and the ZED publish data at the same frequency.
Since IMU can output the inertial frame at 800Mhz, using a floating frame you can reflect the differences between the pose estimated by the localization algorithm and the IMU inertial data.

rhklite, 2019-03-13:
Thank you for the reply! So what might be the impact on pose estimated if this link is set to fixed? And what what might be the impact on odom published by zed if this link is set to fixed?

Myzhar, 2019-03-13:
No impact. That link is used only be eventual node that subscribe to imu topics for elaboration external to ZED node

closed 2019-03-14 by rhklite