stereolabs/zed-ros-wrapper #387: Error saving frame to SVO on Tegra TX1

richrdcm, 2019-03-18:
Hi, I am trying to save a svo file with the rosservice but I get the message:

_Failed to query video capabilities: Inappropriate ioctl for device
NvMMLiteOpen : Block : BlockType = 8 
===== MSENC =====
NvMMLiteBlockCreate : Block : BlockType = 8 
892744264
842091865
[ INFO] [1552918050.746097976]: SVO recording STARTED: hola.svo (HEVC (H265))
[ERROR] [1552918050.789026233]: Error saving frame to SVO
[ WARN] [1552918050.789346850]: Elaboration takes longer than requested by the FPS rate. Please consider to lower the 'frame_rate' setting.
===== NVENC blits (mode: 1) into block linear surfaces =====
[ERROR] [1552918051.820424786]: Error saving frame to SVO

I have SDK 2.7 with CUDA 9.0, and latest master zed-ros-wrapper_

This issue was not happening with SDK 2.6 and the recording svo example, I suppose is the compression type. Is there a way I can change it to LOSSY?

richrdcm, 2019-03-18:
Ok I changed line 2263 of the zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp to LOSSY and worked, could it be a problem with my GPU?
It would be nice to have the compression format as argument on the service too.

Myzhar, 2019-04-01:
The requested features will be available with SDK v2.8

closed 2019-04-01 by Myzhar