stereolabs/zed-ros-wrapper #389: Allow to retrieve and publish point clouds at different rate

bjsowa, 2019-03-20:
I think it would be a good feature to allow publishing point clouds at lower rate than cameras frame rate. 

In my case, I set the frame rate to 30 for positional tracking and use point clouds for obstacle detection. I wrote a Point Cloud Throttle Nodelet to limit outcoming ROS messages to 5hz, but zed wrapper still generates PointCloud2 messages for every camera frame. This results in these warnings:
```
Elaboration takes longer than requested by the FPS rate. Please consider to lower the 'frame_rate' setting
```

Myzhar, 2019-03-25:
@bjsowa since pointcloud publishing runs on a separated thread this feature can be added.
In the future we will add a parameter to set the pointcloud rate.

Myzhar, 2019-04-01:
This feature will be available with SDK v2.8

closed 2019-04-01 by Myzhar