stereolabs/zed-ros-wrapper #390: i have correctly installed zed sdk 2.7 on jetson tx2 with jetpack 3.3 and ROS kinetic has also been installed successfully. when i do 'catkin make' i get the error that 'import yaml' and 'import em' doesnt work.. whereas i have installed both libraries.

agenthimzz, 2019-03-21:
`~/jetsonROS5$ catkin_make
Base path: /home/nvidia/jetsonROS5
Source space: /home/nvidia/jetsonROS5/src
Build space: /home/nvidia/jetsonROS5/build
Devel space: /home/nvidia/jetsonROS5/devel
Install space: /home/nvidia/jetsonROS5/install
####
#### Running command: "make cmake_check_build_system" in "/home/nvidia/jetsonROS5/build"
####
####
#### Running command: "make -j4 -l4" in "/home/nvidia/jetsonROS5/build"
####
[  5%] Built target zed_depth_sub
[ 10%] Built target zed_tracking_sub
[ 15%] Built target zed_video_sub
[ 15%] Built target _zed_wrapper_generate_messages_check_deps_start_svo_recording
[ 15%] Built target _zed_wrapper_generate_messages_check_deps_set_initial_pose
[ 15%] Built target _zed_wrapper_generate_messages_check_deps_stop_svo_recording
[ 15%] Built target _zed_wrapper_generate_messages_check_deps_reset_odometry
[ 17%] Built target zed_wrapper_gencfg
[ 17%] Built target _zed_wrapper_generate_messages_check_deps_reset_tracking
[ 30%] Built target zed_wrapper_generate_messages_nodejs
[ 38%] Generating C++ code from zed_wrapper/stop_svo_recording.srv
[ 38%] Generating Python code from SRV zed_wrapper/stop_svo_recording
[ 48%] Built target zed_wrapper_generate_messages_lisp
[ 51%] Generating C++ code from zed_wrapper/set_initial_pose.srv
[ 66%] Built target zed_wrapper_generate_messages_eus
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/genpy/cmake/../../../lib/genpy/gensrv_py.py", line 43, in <module>
    import genpy.generator
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 44, in <module>
[ 69%] Generating Python code from SRV zed_wrapper/set_initial_pose
    import yaml
ImportError: No module named 'yaml'
zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_generate_messages_py.dir/build.make:66: recipe for target '/home/nvidia/jetsonROS5/devel/lib/python3/dist-packages/zed_wrapper/srv/_stop_svo_recording.py' failed
make[2]: *** [/home/nvidia/jetsonROS5/devel/lib/python3/dist-packages/zed_wrapper/srv/_stop_svo_recording.py] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 71%] Generating C++ code from zed_wrapper/reset_odometry.srv
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 41, in <module>
    import genmsg.template_tools
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 39, in <module>
    import em
ImportError: No module named 'em'
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/genpy/cmake/../../../lib/genpy/gensrv_py.py", line 43, in <module>
    import genpy.generator
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 44, in <module>
    import yaml
ImportError: No module named 'yaml'
zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_generate_messages_cpp.dir/build.make:67: recipe for target '/home/nvidia/jetsonROS5/devel/include/zed_wrapper/stop_svo_recording.h' failed
make[2]: *** [/home/nvidia/jetsonROS5/devel/include/zed_wrapper/stop_svo_recording.h] Error 1
make[2]: *** Waiting for unfinished jobs....
zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_generate_messages_py.dir/build.make:71: recipe for target '/home/nvidia/jetsonROS5/devel/lib/python3/dist-packages/zed_wrapper/srv/_set_initial_pose.py' failed
make[2]: *** [/home/nvidia/jetsonROS5/devel/lib/python3/dist-packages/zed_wrapper/srv/_set_initial_pose.py] Error 1
CMakeFiles/Makefile2:3728: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_generate_messages_py.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_generate_messages_py.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 41, in <module>
    import genmsg.template_tools
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 39, in <module>
    import em
ImportError: No module named 'em'
zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_generate_messages_cpp.dir/build.make:74: recipe for target '/home/nvidia/jetsonROS5/devel/include/zed_wrapper/set_initial_pose.h' failed
make[2]: *** [/home/nvidia/jetsonROS5/devel/include/zed_wrapper/set_initial_pose.h] Error 1
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 41, in <module>
    import genmsg.template_tools
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 39, in <module>
    import em
ImportError: No module named 'em'
zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_generate_messages_cpp.dir/build.make:81: recipe for target '/home/nvidia/jetsonROS5/devel/include/zed_wrapper/reset_odometry.h' failed
make[2]: *** [/home/nvidia/jetsonROS5/devel/include/zed_wrapper/reset_odometry.h] Error 1
CMakeFiles/Makefile2:2879: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_generate_messages_cpp.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_generate_messages_cpp.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`

Myzhar, 2019-03-25:
Hi @agenthimzz ,
I think this is a ROS related error more than a ZED problem. Maybe you have some conflicts with the versions of the libraries like this old ROS issue suggests:
https://github.com/ros/genmsg/issues/63

agenthimzz, 2019-03-26:
thanks.. I found the correct solution from your message.

closed 2019-03-26 by agenthimzz