stereolabs/zed-ros-wrapper #394: ros argument `initial_pose` seems to have no effect

NeilNie, 2019-04-01:
Hello! 

In the `zed_camera.launch` [file](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/launch/zed_camera.launch), I could set the initial position of the camera. However, when I changed this parameters, and then opened rviz to see the camera position, it's still at [0,0,0,0,0,0].

However, if I do a service call and manually set the position, then the camera appears at the correct position. 

I am wondering if there is a bug with the `initial_pose` parameter, or I am doing something wrong. Thanks!

Myzhar, 2019-04-01:
I just tested the behavior of the `initial_pose` parameter and it works as expected.
How are you changing it? Can you post your full launch file to check if it's correct?

Meanwhile I anticipate that the next version of the wrapper, coming with SDK v2.8, will replace the current params method, moving all the parameter settings to dedicated YAML files, so everything will be more clear and easy to be customized (see the branch https://github.com/stereolabs/zed-ros-wrapper/tree/feat_yaml_params).

NeilNie, 2019-04-01:
@Myzhar Thanks for the reply. Here is my launch file: https://github.com/sigmaai/self-driving-golf-cart/blob/3d3e9f0b9f75a33d10cab3df75bb8eca85f4121e/ros/src/sensors/zed-ros-wrapper/zed_wrapper/launch/zed_camera.launch#L52

Using dedicated YAML files would be great! Looking forward to that!

Myzhar, 2019-04-01:
Ok, I found your problem... you have two problems:
**Z value:**
You activated the "Floor detection", so the Z position is updated with the calculated height from the floor:
https://github.com/sigmaai/self-driving-golf-cart/blob/3d3e9f0b9f75a33d10cab3df75bb8eca85f4121e/ros/src/sensors/zed-ros-wrapper/zed_wrapper/launch/zed_camera.launch#L108

The same is for the orientation. However I suggest you to set it to `0` and fix a pitch value in the URDF file, that is the standard approach for ROS

NeilNie, 2019-04-01:
Unfortunately, turning floor detection off didn't change anything. The initial position is still at the origin. 

[image]

This is rviz after I run a service call. 
[image]

Myzhar, 2019-04-01:
I tried on my side and everything works as expected, but I just noticed another problem here:
    `<arg name="initial_pose"                  default="[0.0,0.0,0.9144.0, 0.0,0.52,0.0]" /> `

*0.9144.0* ;-)

NeilNie, 2019-04-01:
I am very sorry to bother you. Everything now works as expected. Thanks a lot for your help!

closed 2019-04-01 by NeilNie