stereolabs/zed-ros-wrapper #397: Sometimes depth image is black in RViz

abylikhsanov, 2019-04-01:
I am running ZED M on Jetson TX1. When I run the camera using ZED Depth viewer I can see depth image quite well but when I run the zed_ros_wrapper and try to visualize the same depth image in RViz, very often I get black image, why is that?

To clarify, I do get depth image in RViz but it interchanges with just a black image.

Myzhar, 2019-04-01:
Hi @abylikhsanov 
I need more information to reply to your question:
* Camera resolution
* Camera framerate
* Where does Rviz run (Jetson TX1 or remote PC)?
* Jetpack installed on the TX1
* Power status of the TX1 (jetson_clocks has been executed?)?

abylikhsanov, 2019-04-01:
Hi @Myzhar 
1. 720p
2. 30 
3. Both
3. 3.3 and L4T 28.2.0
4. No I did not execute jetson_clocks
5. Latest, I believe it is 2.7
6. Latest, just copied the main branch on github

Also, when I simply place the camra flat on the ground I always get black image and if I til a bit I start to see the depth image and if then I move the camera I sometimes get black and sometimes get depth image

Myzhar, 2019-04-01:
If the camera is a ZED the minimum measured depth is 40 cm. If the planar surface is nearest than that distance the depth image will result black, even if it is not facing directly the surface

abylikhsanov, 2019-04-01:
@Myzhar It is not ZED, it is ZED M

Myzhar, 2019-04-01:
With the ZED-M the minimum distance is 20 cm, but the effect is similar:
https://www.stereolabs.com/docs/api/structsl_1_1InitParameters.html#a115c93ba994ec7b5c853375d647a936e

Have you changed the `confidence` parameter?

abylikhsanov, 2019-04-01:
I don't think that this is due to the distance as even if I place camera far behind I still get this issue. I will try to decrease the condifence parameter

abylikhsanov, 2019-04-01:
Yes, it does not look like it is due to the confidence score

abylikhsanov, 2019-04-01:
Hey, I mean it is not due to the confidence  :)

Myzhar, 2019-04-01:
I just noticed a similar issue with the option `normalize_range` enabled. Try to disable it and to set the `Min value` to 0.1 and the `Max value` to 10.0 (or even more if you need a major range)

abylikhsanov, 2019-04-01:
Is that the `zed_wrapper_node` parameter? Can't find it

Myzhar, 2019-04-01:
They are options in the Rviz plugin to visualize images
[image]

abylikhsanov, 2019-04-01:
It indeed made the image stable, than you.
But is that safe to use it that way? In terms of performance

Myzhar, 2019-04-01:
Yes, it's better because Rviz is not forced to parse the whole depth image to recalculate the min and max values for each new frame.
If you know those values it's better to fix them.

Please close the issue if all your doubts are solved.

closed 2019-04-01 by abylikhsanov