stereolabs/zed-ros-wrapper #398: How to use zed_camera_nodelet?

stevemartinov, 2019-04-01:
I want to use zed_camera_nodelet to spawn the camera manager as I want to make a remote mapping using RTABmap.

Here is the example launch file:
```
<launch>
  <include file="$(find freenect_launch)/launch/freenect.launch">
     <arg name="depth_registration" value="True" />
  </include>

  <arg name="rate"  default="5"/>
  <arg name="approx_sync" default="true" /> <!-- true for freenect driver -->

  <!-- Use same nodelet used by Freenect/OpenNI -->
  <group ns="camera">
    <node pkg="nodelet" type="nodelet" name="data_throttle" args="load rtabmap_ros/rgbd_sync camera_nodelet_manager" output="screen">
      <param name="compressed_rate"  type="double" value="$(arg rate)"/>
      <param name="approx_sync"      type="bool"   value="$(arg approx_sync)"/>

      <remap from="rgb/image"       to="rgb/image_rect_color"/>
      <remap from="depth/image"     to="depth_registered/image_raw"/>
      <remap from="rgb/camera_info" to="rgb/camera_info"/>

      <remap from="rgbd_image"      to="rgbd_image"/>
    </node>
  </group>      
</launch>
```

How can I use that launch file to work with ZED M?

Myzhar, 2019-04-01:
You can adapt that example to our example available here:
https://github.com/stereolabs/zed-ros-wrapper/tree/master/examples/zed_nodelet_example

Be sure to remap correctly all the topics

stevemartinov, 2019-04-01:
Okay I see, so will this launch file looks to you correct (without looking at the topic remaps)?
```
<launch>
  <include file="$(find zed_wrapper)/launch/zed_camera_nodelet.launch"/>

  <arg name="rate"  default="30"/>
  <arg name="approx_sync" default="true" /> 

  <group ns="camera">
    <node pkg="nodelet" type="nodelet" name="data_throttle" args="load rtabmap_ros/rgbd_sync zed_nodelet_manager" output="screen">
      <param name="compressed_rate"  type="double" value="$(arg rate)"/>
      <param name="approx_sync"      type="bool"   value="$(arg approx_sync)"/>

      <remap from="rgb/image"       to="rgb/image_rect_color"/>
      <remap from="depth/image"     to="depth_registered/image_raw"/>
      <remap from="rgb/camera_info" to="rgb/camera_info"/>

      <remap from="rgbd_image"      to="rgbd_image"/>
    </node>
  </group>      
</launch>
```

Myzhar, 2019-04-01:
It should be working

stevemartinov, 2019-04-01:
When I launch it I get:
```
waitForService: Service [/zed_nodelet_manager/load_nodelet] has not been advertised, waiting...
```
What service is it waiting for?

Myzhar, 2019-04-01:
Have you started the nodelet manager somewhere?

https://github.com/stereolabs/zed-ros-wrapper/blob/master/examples/zed_nodelet_example/launch/zed_nodelet_laserscan.launch#L40-L41

stevemartinov, 2019-04-01:
After launching the manager I ran into this error, any idea?

`[ERROR] [1554132334.000035034]: The error before refreshing the cache was: According to the loaded plugin descriptions the class rtabmap_ros/rgbd_sync with base class type nodelet::Nodelet does not exist. Declared types are  depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/register image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid realsense2_camera/RealSenseNodeFactory stereo_image_proc/disparity stereo_image_proc/point_cloud2 zed_wrapper/ZEDWrapperNodelet`

Shall I just name the camera nodelet manager as` ZEDWrapperNodelet` with group `zed_wrapper`?

bclaw524, 2022-10-18:
> You can adapt that example to our example available here: https://github.com/stereolabs/zed-ros-wrapper/tree/master/examples/zed_nodelet_example
> 
> Be sure to remap correctly all the topics

Is there an updated link?

closed 2022-10-18 by Myzhar