stereolabs/zed-ros-wrapper #404: record imu with SVO

vivekalgorythma, 2019-04-03:
According to the article here 
https://support.stereolabs.com/hc/en-us/articles/360009986754-How-to-convert-SVO-files-to-AVI-depth-or-image-sequence-
the SVO format does support recording of the IMU messages. I have tried ZED Explorer and the ros service call to record SVOs and none of them contain IMU data. 
The only thing in the imu/data topic is the orientation which is calculated from the output of the VIO. Is there a way to record and replay the data

Myzhar, 2019-04-03:
IMU data cannot be recorded with the SVO.

You can record a ROS bagfile including the IMU topic

vivekalgorythma, 2019-04-03:
@Myzhar but then there will be synchronization issues if we are trying to use the IMU data with the recorded SVO, is there away for solving this?

Alternatively, we can record the compressed images with the IMU data in a rosbag, but what would be the best way to re-construct the point cloud from the recorded data?

Myzhar, 2019-04-03:
You can use `message_filters` to synchronize topics: http://wiki.ros.org/message_filters

vivekalgorythma, 2019-04-03:
Let me re-phrase the question. We would like to record data from zed camera, and evaluate different SLAM algorithms offline. For this to work we need a comprehensive data set that includes IMU data, stereo pair images, and the pointcloud (depth registered). 
if we record the SVO and a rosbag with IMU the when playing them back we can't guarantee that they will start playing at the same time.

Is there a way to generate the the pointcloud from an ros bag without using SVO.

Myzhar, 2019-04-03:
Yes, you can always generate a pointcloud from a depth image using the `depth_image_proc` package: http://wiki.ros.org/depth_image_proc

It is exactly the same used by Rviz to generate the colored point cloud from the couple "RGB" + "Depth"

Myzhar, 2019-04-08:
@vivekalgorythma can you please close the issue if your problem is solved?

closed 2019-07-10 by Myzhar