stereolabs/zed-ros-wrapper #405: How to turn off the ZED odom?

stevemartinov, 2019-04-04:
Hi,

How can I completely turn off the odom produced by zed_wrapper_node?

Myzhar, 2019-04-04:
Hi @stevemartinov 
the question requires more information:
- do you want to turn off Positional Tracking module or do you want to stop the publishing of the "odom" topic?
- do you want to turn off the topics or the TF publishing?

stevemartinov, 2019-04-04:
I just want to stop publishing the Odom topic

Myzhar, 2019-04-04:
Simply do not subscribe it. All the topics are published only if there is at least a subscriber

stevemartinov, 2019-04-04:
But I can't, I need "odom" topic as I am producing it from my encoders and zed publishing to the same topic

Myzhar, 2019-04-04:
ZED publishes the same message, but the topic is different. I'm pretty sue that the encoder puslishes something different from `/zed/odom`.

Are you sure that you are not speaking about TF?

abylikhsanov, 2019-04-04:
@stevemartinov 
There is a parameter called "odometry_topic", just set it to any other name:
```
<param name="odometry_topic" type="string" value="zed_odom"/>
```

abylikhsanov, 2019-04-04:
@Myzhar The default launch file publishes the odometry to ```odom```, here:
https://github.com/stereolabs/zed-ros-wrapper/blob/111725ad1487c1aef387f5ec767e687581c0b7bc/zed_wrapper/launch/zed_camera.launch#L140

stevemartinov, 2019-04-04:
@abylikhsanov 
Thanks, that helped.

closed 2019-04-04 by stevemartinov

Myzhar, 2019-04-04:
The scope of that parameter is exactly that.
However the ZED node (unless you changed the launch file) runs in the `zed` namespace.
Namespaces in ROS are used to be sure that topic name doesn't conflict.
In your case if you publish a `/odom` topic, you can publish also a `/zed/odom` topic (without changing the topic name) and you will not have any kind of conflict

stevemartinov, 2019-04-04:
@Myzhar Yes I know but the default launch file does not appear to group the node with a namespace

Myzhar, 2019-04-04:
Are you using the latest `master` version?

The `zed.launch` file starts the ZED node in the `zed` namespace:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/launch/zed.launch#L38

milan-r-shah, 2019-10-17:
> Hi @stevemartinov
> the question requires more information:
> 
> * do you want to turn off Positional Tracking module or do you want to stop the publishing of the "odom" topic?
> * do you want to turn off the topics or the TF publishing?

Hi,
Could you please tell me how can I turn off TF publishing?

Myzhar, 2019-10-17:
Hi @milan-r-shah 
the ZED TF tree respects the ROS specifications, to get more details about `map`, `odomo` and their meanings you can read the [REP105](https://www.ros.org/reps/rep-0105.html) official documentation that "specifies naming conventions and semantic meaning for coordinate frames of mobile platforms used with ROS."

milan-r-shah, 2019-10-31:
Hi @Myzhar 
Thanks a lot for the quick response and sharing the link of REP105. It was quite helpful.
For those who are also facing this problem, there are two (or at least, that's what I know of) ways to resolve that:

> 1. If the wrapper is being launched by **_zed.launch_** file then it uses one of the parameters files, called **_common.yaml_**. In that file, there are two parameters:
>    - **publish_tf** (publishes TF between odom and base_link)
>    - **publish_map_tf** (publishes TF between map and odom)
> By default, their values are **_true_**. Upon changing them to **_false_**, I was able to resolve the problem.
>
> 2. Another quick solution is to use **_zed_no_tf.launch_** file rather than zed.launch. As its name suggests, it doesn’t publish those 2 TFs and so no need to modify the parameter file.

Myzhar, 2019-10-31:
That's correct.
Please close the issue if you think it's solved.

Don't hesitate to open new issues if you need further help

milan-r-shah, 2019-10-31:
@Myzhar thanks for such quick response. I appreciate that.
By the way, I didn't create any new issue. I think I just commented in one of the existing 'closed' issues. So, to confirm, I don't have to do anything now, right?