stereolabs/zed-ros-wrapper #406: ZED depth estimation correction?

stevemartinov, 2019-04-04:
Hi,
When I try to do mapping I get a map where textureless walls are very thick and deformed. This is due to the ZED algorithm that make featureless walls to be deformed. Have a look at this image (Look how the wall on the left and the ceiling are deformed):
[image]

Are there some parameters tuning that could be done on ZED camera driver to make it interpolate less the textureless areas?

Myzhar, 2019-04-04:
The ZED Camera is a pure stereo camera and has the limitation of the 3D stereo technology.

If a surface is poor in visual information (e.g. a wall with uniform color) the stereo algorithm has no data to be able to reconstruct it in a reliable way.

To get information on how a point of the depth image (and then of the pointcloud) is good you can analyze the Confidence Map.

Myzhar, 2019-04-04:
Moreover to filter points that have a not good confidence you can act on the `confidence` dynamic param:

https://www.stereolabs.com/docs/ros/zed_node/#dynamic-parameters

stevemartinov, 2019-04-04:
How can I analyze confidence map?

Myzhar, 2019-04-05:
You must write your node that subscribes to that topic and to other topics you need (for example depth image). All the matrices are registered, so a pixel of one matrix correspond to the same pixel of all the other matrices (RGB, Depth, confidence, disparity, ...).

https://www.stereolabs.com/docs/depth-sensing/advanced-settings/#confidence-map

stevemartinov, 2019-04-08:
Hi, but isn't the wrapper already publishes the confidence map? black and yellow image

Myzhar, 2019-04-08:
Yes, the wrapper publishes it... but it's to you to use its information to perform your elaborations.
I only linked the documentation that explains what kind of information's coming with the Confidence Map

closed 2019-07-10 by Myzhar