stereolabs/zed-ros-wrapper #408: Issues with /tf message publishing

DRAEYE, 2019-04-06:
Hello, when I run the launch file zed.launch, the /tf message didn't be published correctly, and the warning showed that "The tf from 'zed_left_camera_frame' to 'zed_camera_center' does not seem to be available, will assume it as identity! Verify that the static transforms from URDF are correctly published", but when run the display.launch, it was published well?

Myzhar, 2019-04-06:
If the warning is present only when the node starts you can ignore it, you have no errors.
It is due to a timing problem. In the next release of the wrapper it will not be present anymore

DRAEYE, 2019-04-06:
But the /tf message didn't update over time when run zed.launch

Myzhar, 2019-04-06:
How are you checking it?

DRAEYE, 2019-04-06:
I just type rostopic echo /tf

Myzhar, 2019-04-06:
That's the wrong way. 

To verify that TF are broadcasted correctly you must use other methods:
* tf_echo
* tf_monitor
* view_frames
* rviz
* roswtd

Here you can find the description for each method:
http://wiki.ros.org/tf/Debugging%20tools

DRAEYE, 2019-04-06:
OK, I'll try it, thank you very much.

closed 2019-04-06 by DRAEYE