stereolabs/zed-ros-wrapper #409: Disparity image should have positive values

fujimo-t, 2019-04-11:
It seems that published disparity image has negative values.

Usually, [stereo_msgs/DisparityImage](http://docs.ros.org/lunar/api/stereo_msgs/html/msg/DisparityImage.html) has positive values.

I think #353 is related to this issue and also DisparityImage.min/max_disparity should has positive value.

fujimo-t, 2019-04-11:
I try to fix this.

fujimo-t, 2019-04-13:
Sorry, could you tell me why this is labeled invalid?
If there is something wrong or lack of information, I will close or give a reaction.

Myzhar, 2019-04-13:
The disparity map is not an image, so it's not correct to say that it cannot have negative values.
In our specific case the model used has only negative values and positive values (>0) are garbage generated by noise.

We are used to see disparity maps with positive values because to visualize it as an image we add to each point the abs of the minimum value and the result is then normalized.

fujimo-t, 2019-04-13:
Thank you for explanation, I understand.
Maybe my words were not enough.

Disparity image I said is just stereo_msgs/DisparityImage, not visualized image.
[Specification of DisparityImage](http://docs.ros.org/melodic/api/stereo_msgs/html/msg/DisparityImage.html) expects positive values (left x - right x) at valid region.

```
# Floating point disparity image. The disparities are pre-adjusted for any
# x-offset between the principal points of the two cameras (in the case
# that they are verged). That is: d = x_l - x_r - (cx_l - cx_r)
sensor_msgs/Image image
```

I think that users (at least me) want to ros messages according to specification of stereo_msgs/DisparityImage because this is ROS wrapper.
> In our specific case the model used has only negative values and positive values (>0) are garbage generated by noise.

Myzhar, 2019-04-13:
We work to respect all the ROS specifications and in the description of the DisparityImage topic I do not read that it must have only positive values.

The message description speaks about adjusting x-offset due to vergence,  that is when cameras has not parallel focal axis. But this does not mean that values must be positive.

Even the REP118 does not speak about negative/positive values:
http://www.ros.org/reps/rep-0118.html

As support on what I'm saying you can watch at the description of the dynamic parameters of the `stereo_image_proc` package of the `image_pipeline` stack:
https://github.com/ros-perception/image_pipeline/blob/melodic/stereo_image_proc/cfg/Disparity.cfg#L22

`Disparity to begin search at, pixels (may be negative)`

fujimo-t, 2019-04-13:
I see.
Thank you for taking your time and I will close this.

But then why frame_id of DisparityImage is same to left camera link?
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L97
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/urdf/zed.urdf#L42

I think frame_id of DisparityImage is normally same to optical frame of left camera.
Also In this case baseline is minus value so I think it needs special coordinate whose x axis is inversed from the optical frame.

closed 2019-04-13 by fujimo-t