stereolabs/zed-ros-wrapper #411: Can ros wrapper using svo file publish svo recorded time?

ZhenghaoFei, 2019-04-14:
Hi,

Is there any proper way to let ros wrapper running with svo file to publish svo record timestamp than the current system timestamp? 
I tried to modified all t = ros::Time::now() in mSvoMode to sl_tools::slTime2Ros(mZed.getTimestamp(sl::TIME_REFERENCE_CURRENT)) and it seems work. But I am not sure if this change will break other parts. 

Thanks,

Myzhar, 2019-04-15:
This is a good question. 

I tried to solve this problem a few months ago, but I have not found a good solution.
What you are saying is correct if only the ZED node is running, but ROS is an ecosystem and the time should be coherent for each running node.
 
The good way to work should be to create a ROS clock server and to publish the ZED timestamp as `/clock` message, so any other node can use it as simulation time: http://wiki.ros.org/Clock

Here comes the problem: the ZED wrapper is a nodelet and only when the nodelet starts we can start the clock server, but to start the nodelet we need a nodelet server which requires a `/clock` message to start. It's a snake biting his tail...

There is an open question on ROS answers about this problem, but no reply has come:
https://answers.ros.org/question/301128/clock-server-as-nodelet/

ZhenghaoFei, 2019-04-15:
Thank you Myzhar for your quick reply!  Yes, it is a snake biting his tail problem with the current ROS nodelet... 
Should I close this issue or you prefer to keep it as an open feature request?

Myzhar, 2019-04-15:
Do not close it. It's a reminder 👍

closed 2022-04-19 by github-actions[bot]