stereolabs/zed-ros-wrapper #412: Convert ZED Depth to Point Cloud

angeliquetaylor, 2019-04-16:
Hello,

I collected a dataset using my ZED sensor a couple of years ago before it was supported with Robot Operating System (ROS). I did not record .SVO files because they were not useful at the time, or at least I didn't know how to use them or convert them to the format that I needed. My data was recorded using the ZED SDK. 

Now, that I have processed my data and annotated it for my research, I pretty much have no choice but to continue using it. However, the RGB and Depth data that I recorded with ZED does not immediately translate to ROS topics.

For example, my RGB data works fine when I put it in a rosbag, but my depth data is not working (maybe because it is stereo instead of traditional depth).

Is there any way to reverse engineer my data such that it can be used in ROS? More specifically, I wrote my data to a rosbag, but the depth data is giving me trouble. I assume it is a formatting issue.

Can you all help me with this?

Myzhar, 2019-04-16:
Hi, I need more information about your depth data.
What do you mean with "it is stereo and not a traditional depth"?

Normally you can have two kind of data type in ROS for depth: floating point data and uint16 data (OpenNI compatible).

angeliquetaylor, 2019-04-16:
What is mean is that my depth data does not look like the depth data for a Xtion for example. In other words, I don't think my data is openni compatible.

My depth data is recorded in png format. Is there any way to convert it to a format that can be used in ROS?

Myzhar, 2019-04-16:
Xtion data is in openni format, so each pixel is 16 bit unsigned and the measure unit is in millimeters.

The PNG images can have different data type, so you should open them with an image editor (eg Gimp) to get that information.

angeliquetaylor, 2019-04-16:
Are you sure that GIMP provides the image pixel data type? I'm trying to get the pixel size in gimp but it doesn't seem to be an option: https://docs.gimp.org/en/gimp-image-properties.html

angeliquetaylor, 2019-04-16:
I used Matlab instead. The depth data is uint8

Myzhar, 2019-04-16:
The images have been rescaled and converted to 8bit before saving, so you lost the depth information.
You could try to rescale back to 16 bit (255 -> 65535), but the pointcloud will result not detailed with Z coordinate presenting big steps.

closed 2019-07-10 by Myzhar