stereolabs/zed-ros-wrapper #413: ZED camera frame tf issue - `zed_rtabmap.launch` example point cloud appears to be tilted

NeilNie, 2019-04-19:
Hello,

I was experimenting with the `zed_rtabmap.launch` in the `./examples` directory. I realized the point cloud that I see in the rtabmap visualization tool appears to be tilted. I am wondering if there is a zed tf issue. Here are two screenshots demonstrating the issue. 

[image]
[image]

Thank you very much!

Neil

NeilNie, 2019-04-22:
A quick update. I realized that this issue was caused by the position of my camera. My camera mount is tilted, therefore, making the point cloud tilted as well. 

The issue was resolved by making a new camera mount that is parallel to the ground. 

My new question is, if I would like to use a tilted camera mount, how do I transform the point cloud so that it's in the right orientation relative to the `base_frame` of the robot?

Myzhar, 2019-04-22:
@NeilNie I was just going to ask you if your camera was tilted.
There is a parameter to be activated when the camera is tilted: `camera_flip`
https://www.stereolabs.com/docs/ros/zed_node/#zed-parameters

If you are using the latest version of the wrapper you can find it in `common.yaml` otherwise you must modify its value in `zed_camera.launch`

NeilNie, 2019-04-22:
@Myzhar Thank you very much for the answer. Closing for now.

nathangeorge1, 2019-05-03:
@Myzhar I had a similar question, in that I want to transform the point cloud so that it's in the right orientation relative to the base_frame of the robot, but changing `camera_flip` only inverts the image data, but my point cloud is like 45 degrees off because that's the angle of my camera.

What should I do?

NeilNie, 2019-05-03:
@nathangeorge1 Reopening this issue.

Myzhar, 2019-05-03:
@nathangeorge1 you must modify the URDF introducing a rotation of 45° for the `zed_camera_center` frame respect to the `base_link` frame:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/urdf/zed.urdf#L33

Myzhar, 2019-05-03:
@nathangeorge1 when you find a closed issue and your question is not strictly related to it please open a new issue. This can be more useful for other users.

nathangeorge1, 2019-05-03:
@Myzhar sounds great; I will change the URDF and update here, and I will also open a new issue next time. (I thought it was the exact same issue but it makes sense to just open another issue if the answer was already found.)

NeilNie, 2019-05-03:
@Myzhar sounds like the issue has been resolved. Closing for now.

closed 2019-05-03 by NeilNie