stereolabs/zed-ros-wrapper #414: SDK 2.8 ROS - No camera or pointcloud data

Rafcin, 2019-04-19:
Both the RGB topic and the pointcloud don't work after updating the Zed SDK to 2.8 on the Jetson Xavier. When launching the display example ROS outputs the following errors.

`Error:   Received an invalid TCPROS header.  Each line must have an equals sign.
         at line 103 in /tmp/binarydeb/ros-melodic-cpp-common-0.6.12/src/header.cpp
Error:   Received an invalid TCPROS header.  Each line must have an equals sign.
         at line 103 in /tmp/binarydeb/ros-melodic-cpp-common-0.6.12/src/header.cpp
`

and

`[ERROR] [1555700711.870732203]: Header from subscriber did not have the required elements: md5sum, topic, callerid
`

Rafcin, 2019-04-19:
Solution provided by Myzhar: Uncheck 'Unreliable' in RVIZ

closed 2019-04-19 by Rafcin

Myzhar, 2019-04-19:
It's a known problem since "Lunar" on armhf architectures:
https://answers.ros.org/question/291076/udpros-communication-fails-at-armhf/

The solution is to uncheck `ureliable` for each topic that does not work to force TCP protocol instead of UDP