stereolabs/zed-ros-wrapper #415: error in catkin_make

pigluss, 2019-04-24:
I don't know where the problem is.I hope to get your help！！！
when  catkin_make in my workspace  /home/catkin_zed 

#### Running command: "make -j6 -l6" in "/home/ll/catkin_zed/build"
####
Scanning dependencies of target _zed_wrapper_generate_messages_check_deps_toggle_led
Scanning dependencies of target zed_tracking_sub
Scanning dependencies of target _zed_wrapper_generate_messages_check_deps_reset_odometry
Scanning dependencies of target zed_depth_sub
Scanning dependencies of target _zed_wrapper_generate_messages_check_deps_stop_remote_stream
Scanning dependencies of target zed_video_sub
[  5%] Building CXX object zed-ros-wrapper/tutorials/zed_video_sub_tutorial/CMakeFiles/zed_video_sub.dir/src/zed_video_sub_tutorial.cpp.o
[  5%] Building CXX object zed-ros-wrapper/tutorials/zed_depth_sub_tutorial/CMakeFiles/zed_depth_sub.dir/src/zed_depth_sub_tutorial.cpp.o
[  5%] Building CXX object zed-ros-wrapper/tutorials/zed_tracking_sub_tutorial/CMakeFiles/zed_tracking_sub.dir/src/zed_tracking_sub_tutorial.cpp.o
[  5%] Built target _zed_wrapper_generate_messages_check_deps_reset_odometry
[  5%] Built target _zed_wrapper_generate_messages_check_deps_stop_remote_stream
[  5%] Built target _zed_wrapper_generate_messages_check_deps_toggle_led
Scanning dependencies of target _zed_wrapper_generate_messages_check_deps_start_remote_stream
Scanning dependencies of target _zed_wrapper_generate_messages_check_deps_set_pose
Scanning dependencies of target _zed_wrapper_generate_messages_check_deps_start_svo_recording
[  5%] Built target _zed_wrapper_generate_messages_check_deps_start_remote_stream
[  5%] Built target _zed_wrapper_generate_messages_check_deps_set_pose
[  5%] Built target _zed_wrapper_generate_messages_check_deps_start_svo_recording
Scanning dependencies of target _zed_wrapper_generate_messages_check_deps_set_led_status
Scanning dependencies of target _zed_wrapper_generate_messages_check_deps_reset_tracking
Scanning dependencies of target _zed_wrapper_generate_messages_check_deps_stop_svo_recording
[  5%] Built target _zed_wrapper_generate_messages_check_deps_reset_tracking
[  5%] Built target _zed_wrapper_generate_messages_check_deps_stop_svo_recording
Scanning dependencies of target zed_wrapper_gencfg
[  5%] Built target _zed_wrapper_generate_messages_check_deps_set_led_status
[  6%] Generating dynamic reconfigure files from cfg/Zed.cfg: /home/ll/catkin_zed/devel/include/zed_wrapper/ZedConfig.h /home/ll/catkin_zed/devel/lib/python2.7/dist-packages/zed_wrapper/cfg/ZedConfig.py
Scanning dependencies of target zed_wrapper_generate_messages_py
Scanning dependencies of target zed_wrapper_generate_messages_lisp
[  8%] Generating Python code from SRV zed_wrapper/set_pose
[ 10%] Generating Lisp code from zed_wrapper/set_pose.srv
[ 11%] Generating Lisp code from zed_wrapper/start_svo_recording.srv
Generating reconfiguration files for Zed in zed_wrapper
[ 13%] Generating Lisp code from zed_wrapper/reset_tracking.srv
Wrote header file in /home/ll/catkin_zed/devel/include/zed_wrapper/ZedConfig.h
[ 15%] Generating Lisp code from zed_wrapper/set_led_status.srv
[ 15%] Built target zed_wrapper_gencfg
Scanning dependencies of target zed_wrapper_generate_messages_nodejs
[ 16%] Generating Lisp code from zed_wrapper/stop_svo_recording.srv
[ 18%] Generating Javascript code from zed_wrapper/set_pose.srv
[ 20%] Generating Python code from SRV zed_wrapper/start_svo_recording
[ 22%] Generating Javascript code from zed_wrapper/start_svo_recording.srv
[ 23%] Generating Lisp code from zed_wrapper/stop_remote_stream.srv
[ 25%] Generating Lisp code from zed_wrapper/toggle_led.srv
[ 27%] Generating Javascript code from zed_wrapper/reset_tracking.srv
[ 28%] Generating Python code from SRV zed_wrapper/reset_tracking
[ 30%] Generating Lisp code from zed_wrapper/start_remote_stream.srv
[ 32%] Generating Javascript code from zed_wrapper/set_led_status.srv
[ 33%] Generating Lisp code from zed_wrapper/reset_odometry.srv
[ 35%] Generating Javascript code from zed_wrapper/stop_svo_recording.srv
[ 37%] Generating Python code from SRV zed_wrapper/set_led_status
[ 37%] Built target zed_wrapper_generate_messages_lisp
[ 38%] Generating Python code from SRV zed_wrapper/stop_svo_recording
[ 40%] Generating Javascript code from zed_wrapper/stop_remote_stream.srv
[ 42%] Generating Javascript code from zed_wrapper/toggle_led.srv
[ 44%] Generating Python code from SRV zed_wrapper/stop_remote_stream
[ 45%] Generating Javascript code from zed_wrapper/start_remote_stream.srv
[ 47%] Generating Python code from SRV zed_wrapper/toggle_led
[ 49%] Generating Javascript code from zed_wrapper/reset_odometry.srv
[ 50%] Generating Python code from SRV zed_wrapper/start_remote_stream
[ 52%] Generating Python code from SRV zed_wrapper/reset_odometry
[ 52%] Built target zed_wrapper_generate_messages_nodejs
Scanning dependencies of target zed_wrapper_generate_messages_eus
[ 54%] Generating EusLisp code from zed_wrapper/set_pose.srv
[ 55%] Generating EusLisp code from zed_wrapper/start_svo_recording.srv
[ 57%] Generating EusLisp code from zed_wrapper/reset_tracking.srv
[... 85 lines cut ...]
         sl::ERROR_CODE err = mZed.enableStreaming(params);
                                   ^
/home/ll/catkin_zed/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘bool zed_wrapper::ZEDWrapperNodelet::on_stop_remote_stream(zed_wrapper::stop_remote_stream::Request&, zed_wrapper::stop_remote_stream::Response&)’:
/home/ll/catkin_zed/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:2970:18: error: ‘class sl::Camera’ has no member named ‘disableStreaming’
             mZed.disableStreaming();
                  ^
/home/ll/catkin_zed/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘bool zed_wrapper::ZEDWrapperNodelet::on_set_led_status(zed_wrapper::set_led_status::Request&, zed_wrapper::set_led_status::Response&)’:
/home/ll/catkin_zed/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:2990:32: error: ‘CAMERA_SETTINGS_LED_STATUS’ is not a member of ‘sl’
         mZed.setCameraSettings(sl::CAMERA_SETTINGS_LED_STATUS, req.led_enabled ? 1 : 0);
                                ^
/home/ll/catkin_zed/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘bool zed_wrapper::ZEDWrapperNodelet::on_toggle_led(zed_wrapper::toggle_led::Request&, zed_wrapper::toggle_led::Response&)’:
/home/ll/catkin_zed/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:3002:45: error: ‘CAMERA_SETTINGS_LED_STATUS’ is not a member of ‘sl’
         int status = mZed.getCameraSettings(sl::CAMERA_SETTINGS_LED_STATUS);
                                             ^
/home/ll/catkin_zed/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:3004:32: error: ‘CAMERA_SETTINGS_LED_STATUS’ is not a member of ‘sl’
         mZed.setCameraSettings(sl::CAMERA_SETTINGS_LED_STATUS, new_status);
                                ^
zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/build.make:86: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o' failed
make[2]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o] Error 1
CMakeFiles/Makefile2:2159: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j6 -l6" failed
ll@ll123:~/catkin_zed$

Myzhar, 2019-04-24:
Have you installed the latest SDK v2.8?

If yes then try to delete the folders `devel` and `build` in your catkin workspace and compile again

pigluss, 2019-04-24:
no, I use SDK V2.7 . I am going to try to update my SDK now.

pigluss, 2019-04-24:
Thx, I have sloved this problem after I update my SDK v2.7 to v2.8.

closed 2019-04-24 by pigluss