stereolabs/zed-ros-wrapper #418: one of multi camera always in use

xiaodongzhao, 2019-05-02:
Hi, 

I have two ZED camera connected to the same computer, with only one GPU. 
When I tried to run:
`roslaunch zed_wrapper zed_multi_cam.launch `

I always get this error for one camera:
`ZED >> ERROR: The camera is already used by another process.`

I have tried to reconnect the camera, swap the USB ports, and reboot my computer. 
I have also tried to connect one camera at a time, which is fine using the "zed_camera.launch" file.
However, when I connect two cameras, one of them is always in use. 

Any idea what is wrong? 
Thanks

Myzhar, 2019-05-02:
Hi @xiaodongzhao,

I need more information to understand the cause of your problem:
1. the machine you are using (desktop/laptop, Nvidia Jetson, ...)
2. the version of the ZED SDK installed on the machine
3. the version of the ZED Wrapper used for your tests
4. the type of cameras your are using (ZED+ZED, ZED+ZEDM, ZEDM+ZEDM)

xiaodongzhao, 2019-05-02:
Hi @Myzhar, 

Sorry for missing information. 

1. the machine you are using (desktop/laptop, Nvidia Jetson, ...)
desktop with Ubuntu 18.04, Nvidia 2080Ti GPU

2. the version of the ZED SDK installed on the machine
2.8

3. the version of the ZED Wrapper used for your tests
2.6.0, clone latest from https://github.com/stereolabs/zed-ros-wrapper.git

4. the type of cameras your are using (ZED+ZED, ZED+ZEDM, ZEDM+ZEDM)
ZED+ZED

thanks

jbadger95, 2019-05-13:
We have the same issue, we have 3 ZED stereo cameras, 2 of which are new, this happens intermittently to all 3. We can be running tests with the camera fine multiple times then without changing anything we relaunch tests and the zed fails saying it’s already in use by another process. After closing all programs and aborting processes we try launching again and the problem persists. Running the diagnostics took the camera test fails. Each of the ZEDs also crash when they are moved rapidly. 

We are using an Nvidia jetson with Ubuntu 16.04, SDK version 2.8, ZED+ZED

Myzhar, 2019-05-13:
@xiaodongzhao  @jbadger95 I must report this issue to our HW engineers because it seems not related to the ROS wrapper.

jbadger95, 2019-05-13:
While you’re reporting it to them, you may want to mention that some serious reinforcement should be made to the wire, any slight jarring causes the ZED to crash. We have gobs of hot glue securing ours now but it is a really poor design choice. Securing the back wire did seem to reduce the frequency of the aforementioned ‘already in use failures.’

Myzhar, 2020-03-23:
Solved with latest SDKs

closed 2020-03-23 by Myzhar

jheanelestrada, 2023-05-16:
Hi. I am encountering the same problem. 

    the machine you are using (desktop/laptop, Nvidia Jetson, ...)
- Jetson Orin
    the version of the ZED SDK installed on the machine
- 3.7
    the version of the ZED Wrapper used for your tests
- zed ros2 wrapper and zed-ros2 example (multi_camera)
    the type of cameras your are using (ZED+ZED, ZED+ZEDM, ZEDM+ZEDM)
-Zed + Zed

It's always choose between the 2 zed cameras to open. I tried to run each zed camera separately and it's working fine but calling the multi_camera launch file, the other one always shows an error of : 

Cannot start camera stream. Make sure your camera is not already used by another process or blocked by firewall.
[zed_wrapper-4] [ZED][INFO] [Init]  Failed to open camera. Trying to re-open for 5 seconds

Myzhar, 2023-05-16:
Hi @jheanelestrada 
please do not reply to a closed issue. Create a new one with your information and add a link to the closed issue as a reference if you need it.