stereolabs/zed-ros-wrapper #419: zed_camera_center tf issue

nathangeorge1, 2019-05-03:
Hello,

I used robot_pose_ekf to combine odometry data from /zed/odom and encoders from my robot into /odom, and I’m trying to get the zed_camera_center tf frame to be synced with /odom, but currently it appears to be synced with /zed/odom still.

If arg odometry_frame=odom in zed_wrapper’s zed_camera.launch, why is this happening?

Thanks,
Nathan

Myzhar, 2019-05-03:
If you updated the wrapper to the latest version you must change your configuration.
According to the changelog, the `odom` and `pose` messages are not referred to the `zed_camera_center` frame, but to the `base_link` frame:
https://www.stereolabs.com/docs/ros/zed_node/#transform-frame

We introduced this modification to make easier to connect the ZED camera to a standard robot configuration.

nathangeorge1, 2019-05-03:
Sounds good. I will either try updating to the latest wrapper version or change the link name to `base_link` and update.

nathangeorge1, 2019-05-06:
Updating to the latest ROS wrapper seemed to help, because now my system is using `base_link`, but the `/move_base/global_costmap/footprint` still "jumps" between `/robot_pose_ekf/odom` and `/zed/zed_node/odom`; do you know what might be causing this?

nathangeorge1, 2019-05-06:
I ended up finding the issue was zed_wrapper and I think move_base were both publishing to `base_link`, making the footprint jump. I simply turned zed's `publish_tf`=false and it works!

closed 2019-05-06 by nathangeorge1