stereolabs/zed-ros-wrapper #420: Obtain corresponding rectified image pixel cordinate from a segmented sensor_msgs::pointcloud2 msg.

Arunsid, 2019-05-03:
Hi,am subscribing to the segmented ground plane which is sensor_msgs::Pointcloud2 cloud,and i have subscribed to corresponding camera_info topics . I noticed that i could use the formular to find the image indices as:

u=Fx* cloud.X/(Z) +Cx
v=Fy* cloud.Y/(Z) +Cy

However i noticed that sensor_msgs::pointcloud2 is a one dimentional array,so getting the correct indices of all the segmented pointcloud to do this transformation to get pixel i find a problem.I have extracted the plane using RANSAC plane fitting.below is the following program,

		PointCloudC::Ptr cloud(new PointCloudC());
		pcl::fromROSMsg(msg, *cloud);
		pcl::PointIndices::Ptr indices(new pcl::PointIndices());
		pcl::PointIndices indices_internal;
		pcl::SACSegmentation<PointC> seg;
		seg.setOptimizeCoefficients(true);

		// Search for a plane perpendicular to some axis (specified below).
		seg.setModelType(pcl::SACMODEL_PERPENDICULAR_PLANE);
		seg.setMethodType(pcl::SAC_RANSAC);
		// Set the distance to the plane for a point to be an inlier.
		seg.setDistanceThreshold(0.2);
		seg.setInputCloud(cloud);

		// Make sure that the plane is perpendicular to Z-axis, 10 degree tolerance.
		Eigen::Vector3f axis;
		axis << 0, 0, 1;
		seg.setAxis(axis);
		seg.setEpsAngle(pcl::deg2rad(10.0));

		// coeff contains the coefficients of the plane:
		// ax + by + cz + d = 0
		pcl::ModelCoefficients coeff;
		seg.segment(indices_internal, coeff);
		*indices = indices_internal;

		if (indices->indices.size() == 0) {
		ROS_ERROR("Unable to find surface.");
		return;
		}
		
		PointCloudC::Ptr original_cloud(new PointCloudC);
		PointCloudC::Ptr subset_cloud(new PointCloudC);
	pcl::ExtractIndices<PointC> extract;
	extract.setInputCloud(cloud);
	extract.setIndices(indices);
	extract.filter(*subset_cloud);

now i try to extract the subset pointcloud (sensor_msgs::pointcloud2 )which have the segmented ground information and lets say i have cam_info,how can i get the corresponding pixels in the rectified image .

void callback(image_transport::Publisher& pub_l, const CameraInfoConstPtr& cameraInfoMsg,const sensor_msgs::PointCloud2ConstPtr& subset_cloud, const sensor_msgs::ImageConstPtr& camera_rectified)
{
....................
.........................
how can i get the pixels of the segmenetd point cloud ?
}

[image]
Is der any mapping done between images to pointcloud in zed API?Could anyone help me to get the Mat image that has only pixels corresponding to the segmented ground.

Myzhar, 2019-05-03:
Hi @Arunsid,
to transform from "world" to "camera" coordinates the only way is to apply the formulas you reported to each point of your sub-pointcloud:
u=Fx* cloud.X/(Z) +Cx
v=Fy* cloud.Y/(Z) +Cy

Arunsid, 2019-05-06:
Hi @Myzhar 
I have tried to obtain the pixel cordinates,but its out of bound as well as wrong.I doubt the value of the pointcloud world values to be wrong.How could i obtain the the world cordiantes value from the sensor_msgs::pointcloud2 msg? 
Following the above code ,i have now indices of the segmented cloud and stored to a pcl::PointXYZRGB point;

_for (size_t i=0; i<indices->indices.size(); ++i) {
int index = indices->indices[i];
const PointC& pt = cloud->points[index];
double fx, fy, cx, cy;
double u,v;
 fx=(cameraInfoMsg->K[0])/1000;
fy=(cameraInfoMsg->K[4])/1000;
cx=cameraInfoMsg->K[2];
cy=cameraInfoMsg->K[5];

u=(fx*pt.x)/pt.z+cx;
v=(fx*pt.y)/pt.z+cy;
std::cout<<" cloud x :" <<pt.x<<" cloud y:"<<pt.y<<" and cloud z:"<<pt.z<<" and         image x:"<<u<< "  and image y:"<<v<<std::endl;_
}
for this i was getting output :
**cloud x :1.11848 cloud y:-1.10647 and cloud z:-0.552887 and         image x:596.781  and image y:380.178**

i have the projection matrix value as P= [688.215,0.0,598.172,0.0,0.0,688.215,378.8006,0.0,0.0,0.0,1.0,0.0]

do u know whether i need to convert  sensor_pointcloud2 to world X,Y,Z or its already present in world cordinates ?Do anyone know where i am getting wrong?
[image]

Myzhar, 2019-05-06:
These
x:596.781 and image y:380.178
are you pixel coordinates: u=597, v=380

Why do you say that they are out of bound?

The pointcloud normally is in its own coordinate, referred to the frame indicated in the 'frame_id' field of the sensor_msgs::pointcloud2 msg.
If the image camera and the pointcloud are in the same frame, then [u,v] are the camera coordinates

Arunsid, 2019-05-07:
Hi ,sorry it was mentioned to a case before wehre i just used focal length without conversion to meters.and hence it was out of bound.BUt as per the above program although it seems to be in bound to image aize,the cordinates are either not properly received or are not mapped correctly.

here is the program where inputted the zed pointcloud msg and den used a plane fitting and then tried to remap.however i dont get the right pixel mapped.

sensor_msgs::PointCloud2 my_pcl;
my_pcl = *cloud_msg;

PointCloudC::Ptr cloud(new PointCloudC());
pcl::fromROSMsg(my_pcl, *cloud);
pcl::PointIndices::Ptr indices(new pcl::PointIndices());

pcl::PointIndices indices_internal;
pcl::SACSegmentation<PointC> seg;
seg.setOptimizeCoefficients(true);

// Search for a plane perpendicular to some axis (specified below).
seg.setModelType(pcl::SACMODEL_PERPENDICULAR_PLANE);
seg.setMethodType(pcl::SAC_RANSAC);
// Set the distance to the plane for a point to be an inlier.
seg.setDistanceThreshold(0.2);
seg.setInputCloud(cloud);

// Make sure that the plane is perpendicular to Z-axis, 10 degree tolerance.
Eigen::Vector3f axis;
axis << 0, 0, 1;
seg.setAxis(axis);
seg.setEpsAngle(pcl::deg2rad(10.0));

// coeff contains the coefficients of the plane:
// ax + by + cz + d = 0
pcl::ModelCoefficients coeff;
seg.segment(indices_internal, coeff);

*indices = indices_internal;
if (indices->indices.size() == 0) {
ROS_ERROR("Unable to find surface.");
return;
}

std::vector<cv::Point2i> ground_points;
cv::Mat gray_image;
cv_bridge::CvImagePtr cvimg = cv_bridge::toCvCopy(camera_rectified , sensor_msgs::image_encodings::BGR8);
cv::cvtColor(cvimg->image,gray_image,CV_BGR2GRAY);

for (size_t i=0; i<indices->indices.size(); ++i)
{

	int index = indices->indices[i];
	const PointC& pt = cloud->points[index];
	double fx, fy, cx, cy;
	int u,v;
	 
	fx=(cameraInfoMsg->K[0])/1000;
	fy=(cameraInfoMsg->K[4])/1000;
	cx=cameraInfoMsg->K[2];
	cy=cameraInfoMsg->K[5];

	if (pt.x < 0 || pt.x > camera_rectified->width)
	{
	//ROS_WARN("[PointCloudToImageProjector:] width out of bounds");
	continue;
	}
	if (pt.y < 0 || pt.y > camera_rectified->height)
	{
	//ROS_WARN("[PointCloudToImageProjector:] height out of bounds");
	continue;
	}

	u=(fx*pt.x)/pt.z+cx;
	v=(fy*pt.y)/pt.z+cy;

	std::cout<<" cloud x : " <<pt.x<<" cloud y: "<<pt.y<<" and cloud z: "<<pt.z<<" and     image x: "<<u<< "  and image y: "<<my_pcl.header.frame_id<<v<<std::endl;
	ground_points.push_back(cv::Point2i(u, v));

}

for( unsigned int i = 0; i < ground_points.size(); i++)
{
 gray_image.at<uchar>(Point(ground_points[i].x, ground_points[i].y)) = 255;
}

cvimg->image=gray_image;
cvimg->encoding = "mono8";  
pub_l.publish(cvimg->toImageMsg());

PointCloudC::Ptr original_cloud(new PointCloudC);
PointCloudC::Ptr subset_cloud(new PointCloudC);

pcl::ExtractIndices<PointC> extract;
extract.setInputCloud(cloud);
extract.setIndices(indices);
extract.filter(*subset_cloud);

//std::cout<<"indices "<<*indices<<std::endl;

visualization_msgs::Marker table_marker;
geometry_msgs::Pose pose;
geometry_msgs::Vector3 scale;
Eigen::Vector4f min_pt, max_pt;

pcl::getMinMax3D(*subset_cloud, min_pt, max_pt);

table_marker.pose.position.x = (max_pt.x() + min_pt.x()) / 2;
table_marker.pose.position.y = (max_pt.y() + min_pt.y()) / 2;
table_marker.pose.position.z = (max_pt.z() + min_pt.z()) / 2;
table_marker.pose.orientation.w = 1;

table_marker.scale.x = max_pt.x() - min_pt.x();
table_marker.scale.y = max_pt.y() - min_pt.y();
table_marker.scale.z = max_pt.z() - min_pt.z();

table_marker.ns = "table";
table_marker.header.frame_id = "zed_camera_center";
table_marker.type = visualization_msgs::Marker::CUBE;
table_marker.color.r = 1;
table_marker.color.a = 0.8;

marker_pub.publish(table_marker);

sensor_msgs::PointCloud2 msg_out;
pcl::toROSMsg(*subset_cloud, msg_out);

msg_out.header.stamp = my_pcl.header.stamp;
segment.publish(msg_out);

Arunsid, 2019-05-07:
when i try to visualize i see the mapped image has the shape similar to the segmented points ,i hope the way i plotted the vector in the image is wrong,as i didnt take into consideration of image size.is this the issue?could u tell me how to write this point in code or to take take consideration of the scale?
[image]

Myzhar, 2019-05-07:
This check is wrong, it has no sense:
```
        if (pt.x < 0 || pt.x > camera_rectified->width)
	{
	//ROS_WARN("[PointCloudToImageProjector:] width out of bounds");
	continue;
	}
	if (pt.y < 0 || pt.y > camera_rectified->height)
	{
	//ROS_WARN("[PointCloudToImageProjector:] height out of bounds");
	continue;
	}
```

`pt.x` and `pt.y` are WORLD COODINATES it's correct that their values can be negative and major than image size.

You must check that `u` and `v` obtained here
```
u=(fx*pt.x)/pt.z+cx;
v=(fy*pt.y)/pt.z+cy;
```
are not negative and inside the image margins, u is the horizontal coordinate, v is the vertical coordinate

Arunsid, 2019-05-07:
Yes ur right with that.okay.But my major doubt is now i have vector containing the pointcloud how can map to these pixel and do the rest of the pixel to be zero.Also did you find anything wrong with my pointcloud value from indices of the segmented cloud?Because when i ran now to check with each row,colomn value of image to each of the ground_point vector,only a very few value like 25 row pixel and 6 column pixel output,which gives me doubt whether i am getting the right pointcloud values.

I Have tried now to iterate through vectors to check for image row,clomn as:

std::vector<Point2i>::iterator it;
for (int i = 0; i <= gray_image.rows; i++) {
    for (int j = 0; j <= gray_image.cols; j++) {
      
	it=std::find(ground_points.begin(),ground_points.end(),cv::Point2i(j,i));

		if(it != ground_points.end()){
		std::cout<<"Element found at  "<<i<<"   "<<j<<std::endl;
		}

		else{
		gray_image.at<uchar>(Point(i,j))=255;
		}    
     }
}

Arunsid, 2019-05-07:
[image]
THe present output is i am getting some pixel values,but the number of values is very few and seems something really wrong.

Myzhar, 2019-05-07:
I'm sorry, but this part of your question is not really related to the ZED ROS Wrapper.
It seems more a question to be published on "ROS answers" so you can get a better reply by someone that works daily with point cloud manipulation:
https://answers.ros.org/

Arunsid, 2019-05-08:
HI @Myzhar The pointcloud data is in meters right,so if i want to convert these to image pixels , the camera parametrs for example(focal length and cx,cy)also have to be converted to meters like above equation ?or use the raw camera_info from zed?

Myzhar, 2019-05-08:
The `camera_info` parameters contain the conversion factors, so you must only use the formula we wrote above.

Arunsid, 2019-05-08:
ok.i dont know whether u noticed i have converted the focal length to meters.
fx=(cameraInfoMsg->K[0])/1000;
fy=(cameraInfoMsg->K[4])/1000;
cx=cameraInfoMsg->K[2];
cy=cameraInfoMsg->K[5];
so this input is wrong?

Myzhar, 2019-05-08:
Yes. Camera parameters ARE READY to convert WORLD COORDINATES to CAMERA COORDINATE and viceversa.

Arunsid, 2019-05-08:
okay.in this case i have out of bounds image cordinates are obtained.the image resolution is 1280*720 and also the u,v cordinates are negative.Does this indicate the pointcloud are wrong?
[image]

Myzhar, 2019-05-08:
Can you post the values of Fx, Fy, cx and cy for your camera?

Arunsid, 2019-05-08:
Hi thanks alot.I got the exact output i wanted.yes i didnt take the cordinate systems.I really appreciate your help.
[image]

closed 2019-05-08 by Arunsid