stereolabs/zed-ros-wrapper #421: preventing positional jumping on /zed/odom

nathangeorge1, 2019-05-03:
Hello,

When the ZED camera's image has no features, for example because it's next to a blank wall, the /zed/odom value jumps continuously to various locations in RViz.

After contacting support, they said referring to the ZED-M's IMU, "Preventing positional jumps with the accelerometer is totally possible. However, it's not implemented yet, but it's definitely planned in our tracking roadmap."

I'm curious about:

1. May I know approximately when this could be implemented? I'm trying to get robots to go through hallways.
2. Is there a recommended solution I should take to prevent these positional jumps in the meantime, with or without the IMU?

Thanks,
Nathan

Myzhar, 2019-05-03:
Hi @nathangeorge1

1. As the support says we are working on it, but we have not a fixed timeline for this feature. Stay tuned for news
2. You can use an external Kalman Filter (http://docs.ros.org/melodic/api/robot_localization/html/index.html) that takes ZED-M IMU and `/zed/odom` as inputs and outputs a filtered odometry message that trusts the IMU Accelerations more than the odometry position.

With the latest version of the wrapper you can set a fixed value for odometry covariance, so you can play with the fixed value to give a major weight to IMU data.
See the parameters `fixed_covariance` and `fixed_cov_value` in `common.yaml`:
https://www.stereolabs.com/docs/ros/zed_node/#zed-parameters

nathangeorge1, 2019-05-03:
Great; that makes a lot of sense. I will try that and update here. Would you happen to have a ZED example for robot_localization? I only understand how to use robot_pose_ekf.

Myzhar, 2019-05-03:
There's not an example for the ZED itself, but you find examples that can be adapted on internet

closed 2019-07-10 by Myzhar