stereolabs/zed-ros-wrapper #423: Could NOT find CUDA: Found unsuitable version "9.2", but required is exact version "10.0"

twlee2, 2019-05-04:
I am getting the following issue when I am installing zed-wrapper although I have installed cuda-10.0:

Could NOT find CUDA: Found unsuitable version "9.2", but required is at
  least "10" (found /usr/local/cuda)

Deleting "build" and "devil" folders in workspace causes another issue:

CMake Error at /opt/ros/kinetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/home/user/anaconda3/bin/python
  "/opt/ros/kinetic/share/catkin/cmake/parse_package_xml.py"
  "/opt/ros/kinetic/share/catkin/cmake/../package.xml"
  "/home/user/catkin_ws/build/catkin/catkin_generated/version/package.cmake")
  returned error code 1

Although I have installed catkin_pkg, this issue doesn't go away.

Please advise!

Myzhar, 2019-05-04:
Hi @twlee2,

First of all verify that the default version of CUDA is 10.0 with the command
`$ nvcc -V`

Then you just followed the correct procedure deleting `devel` and `build` folders in your catkin workspace. 

About the error you pasted I have not many info. Have you searched on ROS answers for it?

twlee2, 2019-05-04:
Yes, solved!

Thank you

closed 2019-05-04 by Myzhar

twlee2, 2019-05-04:
But, I am having another issue after installed the zed ros wrapper;

ERROR: cannot launch node of type [zed_wrapper/zed_wrapper_node]: can't locate node [zed_wrapper_node] in package [zed_wrapper]

Can you help with this please?

I did source ~/.bashrc but the issue is still remaining.

Thank you in advance

Myzhar, 2019-05-04:
Try to source your workspace:
`$ source /home/user/catkin_ws/devel/setup.bash`

twlee2, 2019-05-04:
I tried it but the issue still remains..

Myzhar, 2019-05-04:
So the compiling has not succeeded. Try again to launch `$ catkin_make`

mminervini, 2019-05-17:
Hi,

I'm getting a similar, though kind of opposite, error when compiling on Ubuntu 18.04 and ROS Melodic:
```
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find CUDA: Found unsuitable version "10.0", but required is exact
  version "9.0" (found /usr/local/cuda-10.0)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:376 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.10/Modules/FindCUDA.cmake:1080 (find_package_handle_standard_args)
  /usr/local/zed/zed-config.cmake:42 (find_package)
  zed-ros-wrapper/zed_wrapper/CMakeLists.txt:22 (find_package)
```
This happens both with `catkin_make` and `catkin_tools`, and also after deleting `build` and `devel` folders in my workspace.

By the way, installed CUDA is indeed V10.0.130, so I'm surprised that instead CMake is looking for 9.0.

Any advice on how to compile successfully?
Thanks!

Myzhar, 2019-05-17:
@mminervini can you paste the result of the command `$ nvcc -V`?

mminervini, 2019-05-17:
@Myzhar thank you for your quick reply!

Here is the output of `nvcc -V`:
```
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2018 NVIDIA Corporation
Built on Sat_Aug_25_21:08:01_CDT_2018
Cuda compilation tools, release 10.0, V10.0.130
```

P.S. I see that Ubuntu 16.04 and ROS Kinetic are listed in the pre-requisites...are also Ubuntu 18.04 and ROS Melodic supported?

Myzhar, 2019-05-17:
Yes, I'm currently working under Ubuntu 18.04 with ROS Melodic and CUDA10.

`Could NOT find CUDA: Found unsuitable version "10.0", but required is exact
  version "9.0" (found /usr/local/cuda-10.0)`

Which version of ZED SDK are you using? Older version required CUDA9, only latest SDK requires CUDA10

mminervini, 2019-05-17:
I had an older SDK 2.5 (which seemed fine according to ZED SDK ≥ 2.3), so I just installed the latest SDK 2.8 for Ubuntu 18 and CUDA 10.
I tried to run the ZED Diagnostic tool, and it was reporting an error related to CUDA 9.0, which was still on my system. Therefore, I hunted down and removed all CUDA 9 related packages and finally the ZED Diagnostic tool was ok with the SDK.

However, when getting back to compiling the ROS wrapper, after restarting the system and deleting `build` and `devel`, I still got the same error `Could NOT find CUDA: Found unsuitable version "10.0", but required is exact version "9.0" (found /usr/local/cuda-10.0)`.

I'm a bit confused!

Myzhar, 2019-05-17:
Maybe that your system still tries to use the old SDK v2.5.
Check the folder `/usr/local` and delete the folder `old_zed` if it exists.
Than run the command `sudo ldconfig` to update the library cache

mminervini, 2019-05-17:
Everything was built successfully...it worked!
Thank you very much for your help!

Myzhar, 2019-05-17:
@mminervini you are welcome

Myzhar, 2019-05-17:
@twlee2 is your problem solved? Can we close this issue?

mminervini, 2019-05-17:
Yes, sure. I think this issue can be closed.