stereolabs/zed-ros-wrapper #426: Autocalibration

ShreyasSkandanS, 2019-05-13:
Are there any details in the documentation regarding the auto-calibration of the stereo cameras? I'd like to know more about what's going on under the hood and how to use it effectively. I can't seem to find any information in the API docs.

Thanks

Myzhar, 2019-05-13:
This is the link to the API documentation:
https://www.stereolabs.com/docs/api/structsl_1_1InitParameters.html#affeaa06cfc1d849e311e484ceb8edcc5

```
At initialization, the Camera runs a self-calibration process that corrects small offsets from the device's factory calibration.
A drawback is that calibration parameters will sligtly change from one run to another, which can be an issue for repeatability.
If set to true, self-calibration will be disabled and calibration parameters won't be optimized.
```
We suggest to keep the self calibration active, unless for some reason you need to keep the camera parameters `fixed` in different node runnings.

ShreyasSkandanS, 2019-05-14:
Are there details on how this is performed or is this closed to the public?

Myzhar, 2019-05-14:
I'm sorry, but it's part of the not open SDK lib

ShreyasSkandanS, 2019-05-14:
Alright, thank you!

closed 2019-05-14 by ShreyasSkandanS