stereolabs/zed-ros-wrapper #427: Difference in topic /zed/left/camera_info and /zed/left/camera_info_raw

AJwgnr, 2019-05-13:
When using the ZED I have two camera_info topics for each camera:
1. /zed/left/camera_info
2. /zed/left/camera_info_raw

The contents of those topics differ slightly in the D and K matrices.

**/zed/left/camera_info:**

header:
  seq: 268
  stamp:
    secs: 1557745090
    nsecs: 768076514
  frame_id: "zed_left_camera_optical_frame"
height: 720
width: 1280
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [681.6400756835938, 0.0, 684.2488403320312, 0.0, 681.6400756835938, 347.6465148925781, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [681.6400756835938, 0.0, 684.2488403320312, 0.0, 0.0, 681.6400756835938, 347.6465148925781, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False

**/zed/left/camera_info_raw:**
  header:
  seq: 344
  stamp:
    secs: 1557745095
    nsecs: 835038657
  frame_id: "zed_left_camera_optical_frame"
height: 720
width: 1280
distortion_model: "plumb_bob"
D: [-0.1684180051088333, 0.021571900695562363, 0.0, 0.0, 0.0]
K: [701.1580200195312, 0.0, 665.5349731445312, 0.0, 701.1580200195312, 354.885009765625, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [701.1580200195312, 0.0, 665.5349731445312, 0.0, 0.0, 701.1580200195312, 354.885009765625, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False

I have already read https://github.com/stereolabs/zed-ros-wrapper/issues/343#issuecomment-454814788.
Can someone explain me, why there is this difference in the camera info messages?

Myzhar, 2019-05-13:
`camera_info` is referred to rectified images
`camera_info_raw` is referred to unrectified images

Myzhar, 2019-05-13:
I noticed from the name of the topics that you are using an old version of the wrapper.
Please consider to update the ZED wrapper to the latest version to get the latest features and improvements.

AJwgnr, 2019-05-13:
> `camera_info` is referred to rectified images
> `camera_info_raw` is referred to unrectified images

Thanks for the fast clarification!

AJwgnr, 2019-05-13:
> I noticed from the name of the topics that you are using an old version of the wrapper.
> Please consider to update the ZED wrapper to the latest version to get the latest features and improvements.
> 
> Here you an find the documentation of the latest wrapper
> https://www.stereolabs.com/docs/ros/zed_node/

Oh thank you. I will update the wrapper :)

closed 2019-05-13 by AJwgnr