stereolabs/zed-ros-wrapper #428: No camera_info timestamp

abylikhsanov, 2019-05-13:
I have noticed that there is no timestamp on camera_info topic
‘’’
header: 
  seq: 808
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: "zed_right_camera_optical_frame"
height: 720
width: 1280
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [676.9194946289062, 0.0, 638.7445678710938, 0.0, 676.9194946289062, 349.6361999511719, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [676.9194946289062, 0.0, 638.7445678710938, -42.64883804321289, 0.0, 676.9194946289062, 349.6361999511719, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---
‘’’

abylikhsanov, 2019-05-14:
Hi, any update on this issue?

Myzhar, 2019-05-14:
We are working on the patch. Stay tuned

PR #430 (2019-05-20): Fix 428
Fixes #428: added timestamp to `camera_info` messages
closed 2019-05-20 by nesnes