stereolabs/zed-ros-wrapper #432: catkin_make error for ZED wrapper

ThanasisTs, 2019-05-20:
Hello,

I am currently trying to use the ZED wrapper for ROS. Once I download the wrapper and i use the command "catkin_make" I get the following error "No rule to make target '/usr/local/cuda-10.1/lib64/libcublas.so', needed by...". In my PC I have installed both cuda 10.0 and cuda 10.1. I was wondering if the existence of 10.1 affects the system and how that can be fixed. Thanks in advance.

Myzhar, 2019-05-20:
Hi @ThanasisTs 
I need more info to reply to your question:
1. Version of the ZED SDK installed
2. Type of system (desktop, Nvidia Jetson, ...)
3. OS version
4. Reply to the command `$ nvcc -V`

The latest ZED SDK (v2.8) requires CUDA10 (CUDA 10.1 is not supported).

ThanasisTs, 2019-05-20:
Hi @Myzhar 

I have installed ZED SDK 2.8, the type of system is desktop i am using kubuntu 16.04 and the result of the nvcc command is:

nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2015 NVIDIA Corporation
Built on Tue_Aug_11_14:27:32_CDT_2015
Cuda compilation tools, release 7.5, V7.5.17

Myzhar, 2019-05-20:
Well, you have not only CUDA 10 and CUDA 10.1, but CUDA 7.5 too and the latest is your default CUDA version.

What I can suggest is to remove any version of CUDA (check the folder `/usr/local/cuda-*` to verify what you have installed) and install again only the version you are going to use.

ThanasisTs, 2019-05-21:
ok thanks a lot!!

closed 2019-05-23 by ThanasisTs

juanmed, 2019-08-26:
I think the documentation here should be changed then to clarify the need for CUDA 10 and not CUDA10.1 since the links provided in the documentation takes you to the latest CUDA (CUDA10.1) which as mentioned above does not work.

Myzhar, 2019-08-26:
@juanmed can you provide me the link to the wrong documentation?
We changed all the links, but maybe that we missed one.