stereolabs/zed-ros-wrapper #433: static tranform to zed_camera_center

jsseng, 2019-05-23:
I am trying to publish a static transform from 'base_link' to 'zed_camera_center'.  When I run my launch file, the transform sometimes is correct and other times 'base_link' and 'zed_camera_center' are at the same location.  

I have positional tracking turned off.  I am using a Jetson TX2 with SDK v2.8.

Myzhar, 2019-05-23:
Hi @jsseng,
Do you have your own urdf file that publishes the `base_link -> zed_camera_center` transform?
The latest version of the wrapper takes care of it and calculates directly the transform `Odom -> base_link `.
Please paste the result of the command `$ roswtf` to have a better idea of what's happening.

jsseng, 2019-05-23:
I am publishing the following:

``` <node pkg="tf2_ros" type="static_transform_publisher" name="zed_broadcaster" args=".2 0 .5 0 0 0 1 base_link zed_camera_center" />```

When I run the node, I get this:

[ INFO] [1558638236.954703702]: Static transform Sensor to Base [zed_left_camera_frame -> base_link]
[ INFO] [1558638236.954955603]:  * Translation: {0.000,-0.060,0.000}
[ INFO] [1558638236.955043123]:  * Rotation: {0.000,-0.000,0.000}
[ INFO] [1558638236.955181266]: Static transform Sensor to Camera Center
[zed_left_camera_frame -> zed_camera_center]
[ INFO] [1558638236.955243409]:  * Translation: {0.000,-0.060,0.000}
[ INFO] [1558638236.955297841]:  * Rotation: {0.000,-0.000,0.000}
[ INFO] [1558638236.955449103]: Static transform Camera Center to Base [zed_camera_center -> base_link]
[ INFO] [1558638236.955514607]:  * Translation: {0.000,0.000,0.000}
[ INFO] [1558638236.955589966]:  * Rotation: {0.000,-0.000,0.000}

It tells me the static transform from camera to base_link is the same, and this is what I am trying to resolve

Here is roswtf:

================================================================================
Beginning tests of your ROS graph. These may take awhile...
analyzing graph...
... done analyzing graph
running graph rules...
... done running graph rules

Online checks summary:

Found 1 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING The following node subscriptions are unconnected:
 * /zed/zed_state_publisher:
   * /zed/joint_states

Myzhar, 2019-05-23:
The problem is that the ZED Node publishes the same transform reading it from the URDF, so if you want to overwrite it you must modify the URDF.

jsseng, 2019-05-23:
That fixed it.  Thanks for the help!

closed 2019-05-23 by jsseng