stereolabs/zed-ros-wrapper #434: odom twist information

rhklite, 2019-06-12:
i was wondering if in the new Zed SDK v2.8, was twist information implemented?as referenced in #361. 

I could not find any sign of publishing twist in the `zed_wrapper_nodelet.cpp` file

Myzhar, 2019-06-12:
I'm sorry, but it's not yet available since it requires more testing before being released.

MikeWrockAIS, 2019-11-22:
Has there been any progress on this feature? I'm happy to test out a development branch if one is available

Myzhar, 2019-11-22:
Hi @MikeWrockAIS 
not yet sorry. Maybe it will be integrated when the new SDK will be released

rhklite, 2020-02-21:
@Myzhar heyy, i'm wondering if this feature has been integrated into SDK v3.1 yet?

carlos-argueta, 2021-05-06:
Any idea about when twist information will be available? I am using a Zed 2 at the moment.

Myzhar, 2022-07-04:
We are investigating how to use the Sophus (i.e. Lie Algebra SO(3)) library in ROS to correctly transform the odometry twist from the camera frame to the robot base frame. 
The TF2 library allows to transform poses from one frame to another, but not twists.
Updates will come...

ralbra, 2023-03-28:
Hey ist this still worked on?