stereolabs/zed-ros-wrapper #435: PointCloud2 intensity informaiton

arkinrc, 2019-06-19:
Hello,

I am working with ZED camera on nvidia jetson tx2 with ros kinetic and i was confused about PointCloud2 (/zed/zed_node/point_cloud/cloud_registered) generated by zed camera. my question is whether this pointcloud provide the intensity information or not? i checked the nodelet code of zed_ros_wrapper and it seems this point cloud only provide xyzrgb data information. 

1- Is there any possible way that i could get the intensity information of specific area through point cloud generated by above mentioned pointcloud topic? 

2- i also want to get the each point frequency so that i can use those frequencies in my project for other purposes. how can i get these frequencies through pointcloud topic? 

looking forward to your reply and will be thankful your guidance. :)

Myzhar, 2019-06-19:
Hi @arkinrc 
intensity is an information normally related to pointcloud generated by a sensor "laser based" and it's normally related to the reflectivity of the object hit by a laser ray.

The ZED camera is a stereo vision camera and the information available for each pixel is the depth and the color. There is another extracted information that is the "confidence level" related to the depth value of the pixel, I don't know if this can be used as "Intensity" for your algorithms. 

About your second question, can you explain better what you mean with "frequency"?

Myzhar, 2019-07-10:
Hi @arkinrc 
have you find a reply to your question? Can the issue be closed?

arkinrc, 2019-07-17:
Hi @Myzhar 
firstly extremely sorry for very late feedback and thank you so much for your concern related this issue. Yes i have resolved this issue using other sensor such as Sonar & Laser through which i calculated the intensities.

closed 2019-07-17 by arkinrc