stereolabs/zed-ros-wrapper #436: No nvidia graphics card detected

amranov, 2019-06-19:
Hi guys '
i am trying to launch zed_wrapper with `roslaunch zed_wrapper zed.launch `
and i get this error :

```
No NVIDIA graphics card detected. Install an NVIDIA GPU, CUDA and restart your computer after completing installation.
[ INFO] [1560938195.375299478]: ZED connection -> NO GPU DETECTED
in Camera.cpp : void sl::Camera::close() at 163 : cuda error [38]: no CUDA-capable device is detected.
in Camera.cpp : void sl::Camera::close() at 204 : cuda error [38]: no CUDA-capable device is detected.
```

`nvcc -V `output:
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2015 NVIDIA Corporation
Built on Tue_Aug_11_14:27:32_CDT_2015
Cuda compilation tools, release 7.5, V7.5.17
ROS kinetic 16.04 . cuda version 10.0 SKD 2.8.1

Myzhar, 2019-06-19:
Hi @amranov ,
what is the configuration of your system? [CPU, GPU, ...]

amranov, 2019-06-19:
Graphics : Intel® Iris Plus Graphics 650 (Kaby Lake GT3e) 
Processor : Intel® Core™ i7-7567U CPU @ 3.50GHz × 4

Myzhar, 2019-06-19:
The ZED SDK requires and Nvidia card to execute CUDA code, that's why you are getting the message error `No NVIDIA graphics card detected`.
Just to make a double check, can you post the result of the command `$ nvidia-smi`?

amranov, 2019-06-20:
@Myzhar  sorry for late reply 
i got
```
$ nvidia-smi
 nvidia-smi: command not found
```

Myzhar, 2019-06-20:
This confirms that you have not a working Nvidia GPU card.
There are 2 possibilities:
- You have installed a Nvidia GPU card, but the driver is not installed
- You have not installed a Nvidia GPU card

A Nvidia GPU card is a mandatory requirement to use the ZED SDK because it is based on CUDA

closed 2019-07-10 by Myzhar

srinath7979, 2021-07-22:
Hi

I get the following error
./ZED_Tutorial_3 
Error NO GPU DETECTED, exit program.
CUDA error at Camera.cpp:94 code=803(cudaErrorSystemDriverMismatch) "void sl::Camera::close()" 

But i have an nvidia graphics card
nvidia-smi
Failed to initialize NVML: Driver/library version mismatch

Myzhar, 2021-07-22:
Hi @srinath7979 
please never reply to a closed issue. Always open a new one and put a link to the closed as a reference.
In your case the error is self-explaining: there is a mismatch between the Nvidia driver version and the CUDA version. I suggest you remove them all and install the driver and CUDA from scratch.