stereolabs/zed-ros-wrapper #437: Simplify repository

rbonghi, 2019-06-19:
I'm using [wstool](http://wiki.ros.org/wstool) to download quickly a catkin working space for my project and [rosdep](http://wiki.ros.org/rosdep) to install all required dependencies .

After building the _catkin_ws_ I usually start [rosdep](http://wiki.ros.org/rosdep) with this option ([point 2.3.2](http://wiki.ros.org/rosdep#Install_dependency_of_all_packages_in_the_workspace))
```
rosdep install --from-paths src --ignore-src -r -y
```
This "magically" installs all the packages that the packages requires.
But when `rosdep` read all package.xml in this repo, the system start to download all dependencies not only for zed-wrapper but for all examples and tutorials contained in this repository, a significant space for an embedded device (like NVIDIA Jetson Nano).

Is it possibile to split this repository in one with only the zed-ros-wrapper and a secondo with examples and tutorials?

Myzhar, 2019-07-10:
The requested modification is queued and will be implemented in a future release

Myzhar, 2020-02-04:
Done in #509

closed 2020-02-04 by Myzhar