stereolabs/zed-ros-wrapper #438: disable tracking option

jsseng, 2019-06-19:
Is there a way to disable tracking and just use the ZED as a depth camera (to save CPU usage)?  I see the odom topic is still being published.  I disabled publishing the tf.

There is an option to disable mapping in the common.yaml, but I cannot find one to disable tracking.

Myzhar, 2019-06-19:
Hi @jsseng ,
yes, you can. 

You must disable all the features related to positional tracking. Set these parameters in `params/common.yaml`:
`depth_stabilization` -> false
`publish_tf` -> false
`publish_map_tf` -> false

jsseng, 2019-06-19:
I do have those settings set that way, but I still see the odom topic and it is publishing Odometry messages.  Does it still publish the messages or should the 'odom' topic disappear?

Myzhar, 2019-06-19:
The odom topic is still available, but it is not published unless someone subscribes to it.

Disabling the features that I listed above the tracking module is not automatically started, unless odom or pose topics are subscribed.

jsseng, 2019-06-19:
Will doing a 'rostopic echo /zed/odom' start the tracking temporarily then and then it stops when I quit?

Myzhar, 2019-06-19:
No, if the tracking is started  it is not stopped to not lose the position if a disconnection happens due to network problems

jsseng, 2019-06-19:
Okay, that makes sense then.  I noticed my CPU usage would go up 10%, but not back down.  Thank you so much for you help!

closed 2019-06-19 by jsseng

Myzhar, 2019-06-19:
You are welcome. Don't hesitate to contact us if you need more help