stereolabs/zed-ros-wrapper #439: CUDA_npps_LIBRARY not set during catkin_make of 'zed-ros-wrapper'

RatheeshRK, 2019-06-21:
CUDA 9.0 is installed. 
**Error while performing catkin_make of zed-ros-wrapper.**

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_npps_LIBRARY (ADVANCED)
    linked by target "zed_wrapper_node" in directory /home/u16mac/ZED_ws/src/zed-ros-wrapper/zed_wrapper
    linked by target "ZEDWrapper" in directory /home/u16mac/ZED_ws/src/zed-ros-wrapper/zed_wrapper

-- Configuring incomplete, errors occurred!

Myzhar, 2019-06-21:
Hi @RatheeshRK ,
can you provide these information?
- SDK version
- GPU type

Thank you

Myzhar, 2019-07-10:
Hi @RatheeshRK ,
have you solved your problem?

wangq0688, 2020-02-08:
> Hi @RatheeshRK ,
> can you provide these information?
> 
>     * SDK version
> 
>     * ROS wrapper version
> 
>     * GPU type
> 
> 
> Thank you

Hi,I have the same problem. 
sdk 3.0.1
gpu 960m

Myzhar, 2020-03-23:
Closed for inactivity

closed 2020-03-23 by Myzhar