stereolabs/zed-ros-wrapper #440: How can I build my own ZED wrapper? (In hope of improved execution efficiency)

JeyP4, 2019-06-28:
Hi

I am a fan of StereoLab ZED.

Since I need only image and depth messages, I am thinking to build a simple wrapper. I know that _'zed-ros-wrapper'_ publishes on a topic, only if there is a subscriber of it. Still, I am hoping a performance improvement with simple wrapper.(Please let me know, if I am wrong).

Can you guide me how to start with. I am already looking into _['Stereolabs ZED - Examples'](https://github.com/stereolabs/zed-examples)_.
Or may be if you can share any link for simple ZED wrapper. My aim is to use `getTimestamp(TIME_REFERENCE_IMAGE)` also.

Myzhar, 2019-06-28:
Hi @JeyP4 
You can follow the ROS tutorial to create a simple punlisher:
http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28c%2B%2B%29

Than you can add the code to initialize the ZED and to get the data you need following the guidelines of the ZED Example.

Don't hesitate to contact us for further help.

Myzhar, 2019-07-10:
@JeyP4 I close the issue. Don't hesitate to re-open it if you need further help

closed 2019-07-10 by Myzhar