stereolabs/zed-ros-wrapper #441: Catkin_make Error

kartikpaigwar, 2019-06-29:
Cuda 10
Ubuntu 16.04
Zed SDK 2.8.2

Error:

CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:90 (find_package):
  Could not find a package configuration file provided by "roslint" with any
  of the following names:

    roslintConfig.cmake
    roslint-config.cmake

  Add the installation prefix of "roslint" to CMAKE_PREFIX_PATH or set
  "roslint_DIR" to a directory containing one of the above files.  If
  "roslint" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  zed-ros-wrapper/zed_wrapper/CMakeLists.txt:41 (find_package)

-- zed_wrapper: 0 messages, 9 services
CMake Error at zed-ros-wrapper/zed_wrapper/CMakeLists.txt:156 (roslint_cpp):
  Unknown CMake command "roslint_cpp".

-- Configuring incomplete, errors occurred!
See also "/home/kartik/RBC_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/kartik/RBC_ws/build/CMakeFiles/CMakeError.log".
Makefile:1424: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

Myzhar, 2019-06-29:
Hi @kartikpaigwar 

You can install the missing dependency with the command
`sudo apt-get install ros-kinetic-roslint.`

kartikpaigwar, 2019-06-29:
Hi @Myzhar 
 Thanks for resolving the previous error.
 Please help me to resolve the new error about the diagnostic viewer.

n file included from /home/kartik/RBC_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:21:0:
/home/kartik/RBC_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/include/zed_wrapper_nodelet.hpp:41:51: fatal error: diagnostic_updater/diagnostic_updater.h: No such file or directory
compilation terminated.
zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/build.make:86: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o' failed
make[2]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o] Error 1
CMakeFiles/Makefile2:3850: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Myzhar, 2019-06-29:
It's another missing dependency.

To install all the dependency required you can go to the root of your catkin workspace and run this command:
`$ rosdep install --from-paths src --ignore-src -r -y`

Here you can find the documentation about the rosdep command:
 http://wiki.ros.org/rosdep

kartikpaigwar, 2019-06-29:
Thank you so much. I think I need to revise ROS.

closed 2019-06-29 by kartikpaigwar