stereolabs/zed-ros-wrapper #442: Poor Point Cloud, Flat surface gets curved in the point cloud

kartikpaigwar, 2019-07-01:
Hey, I am too getting a very bad point cloud. A flat surface (floor in my case) seems to get curvy. And the whole point cloud looks pretty bad. Does anyone have the same issue as me and found any ways to correct it. Please help. I am using Zed Mini, latest SDK, Cuda 10.
[image]
[image]
[image]

Myzhar, 2019-07-01:
Hi @kartikpaigwar ,

can you provide better screenshoot with the relative RGB images just to better understand in what kind of environment you are working?

Your problem is not easy to be faced with these screenshoots.

Please try also the $ "/usr/local/zed/tools/ZED Depth Viewer" tool and verify that the result is the same

kartikpaigwar, 2019-07-01:
Hi @Myzhar 
These are the stairs point cloud
[image]
. I have attached a few more images for your reference.
[image]
[image]

Myzhar, 2019-07-01:
The surface of the stairs seems quite reflective, that can be the cause of your problems.
Can you please test the ZED Depth Viewer tool and verify that you have the same results?

kartikpaigwar, 2019-07-01:
Yes, I checked the point cloud on depth viewer, what I observed is that Zed camera performs very poor on the shiny surface. The table is not such reflective, though it performs poorly. If we are dealing with the real world problem statement, the surface may not be so perfect. I hope Zed team should find a solution and come up with robust sdk.
[image]

Myzhar, 2019-07-01:
The ZED camera is a pure Stereo Vision device and suffer of the limits of this technology.
Even if the ZED SDK algorithms for surface reconstruction are the most advanced, it is not possible to extract information from a surface that does not have information to be used to extract depth through matching and triangulation.
The uniform homogeneous color of the desk is really poor of visual information and the plane reconstruction is not too bad. 
Have you tried to set depth parameter to ULTRA mode?

kartikpaigwar, 2019-07-01:
Hi @Myzhar ,
Thanks for looking into this issue. Yes, you are correct, it's the stereo vision limitation, can't blame ZED. I tried Ultra mode too, it slightly improves the performance, but can't use the ultra mode on my robot system due to lack of computation.

closed 2019-07-01 by kartikpaigwar