stereolabs/zed-ros-wrapper #443: Tutorials

denzle83, 2019-07-01:
There's a need to update your online instructions for tutorials/samples/examples. Most of the instructions are badly out of date or make very little sense on how to use. Especially for a beginner like myself.

Myzhar, 2019-07-10:
Hi @denzle83 ,
thank you for your suggestion.

Can you provide the links of the instructions you are referring to, so we can improve them?

Meanwhile have you seen the Stereolabs online documentation?
https://www.stereolabs.com/docs/ros/

You can find a full description of the examples:
https://www.stereolabs.com/docs/ros/video/#video-subscribing-in-c
https://www.stereolabs.com/docs/ros/depth_sensing/#depth-subscribing-in-c
https://www.stereolabs.com/docs/ros/position/#position-subscribing-in-c

Myzhar, 2020-03-23:
A new Tutorials/Examples repository is now available:
https://github.com/stereolabs/zed-ros-examples

closed 2020-03-23 by Myzhar