stereolabs/zed-ros-wrapper #444: USB3.0/Resolution error

adiisawant, 2019-07-02:
Hello, I am using ZED camera with Orbitty carrier board on Jetson TX2.

I installed ros-wrapper and tried to run the zed node. I get an error: "ZED >> ERROR: Cannot initialize the camera. Try another resolution."
The default resolution in which it opens is HD720, Framerate:30. When I open the camera in ZED Explorer, it opens in VGA resolution and on trying to change the resolution, it says that USB 3.0 is not detected. Same is the error while running ZED Diagnostics.

I have installed the latest SDK, Linux version is 16.04 with CUDA 9.

Please help.

Thanks!

Myzhar, 2019-07-02:
Hi @adiisawant 
What Jetpack version are you using?
If you correctly connected the ZED to the USB3 port, then it seems a problem with the configuration of your TX2
Be sure to have installed the correct Kernel module for the Orbitty Carrier board:
http://connecttech.com/resource-center/cti-l4t-nvidia-jetson-board-support-package-release-notes/

Myzhar, 2019-07-10:
@adiisawant have you solved your problem?

adiisawant, 2019-08-05:
Hello, I am using Jetpack 3.3. I had also installed the correct Kernel for Orbitty board. The board went defective. However, I am using the same setup with Auvedia J120 Carrier Board now and have flashed the Jetson TX2 with correct Auvedia patch. Issue is still the same. ZED diagnostic says that USB 3.0 is not detected. Also, I cannot run the ZED node through ROS because it says that it didn't find the ZED node. Please assist.

adiisawant, 2019-08-05:
I am following the below repository:

https://github.com/NVIDIA-AI-IOT/redtail/wiki/Skypad-Software-Setup-Guide

Myzhar, 2019-08-05:
Have you tested the ZED on a PC to verify that it correctly works over USB3?
Are you using a hub or have you connected the ZED directly to the USB port?

ArghyaChatterjee, 2019-11-03:
I was also facing the same issue. May be this issue is due to Jetson TX2 supports only one USB 3.0 port (compared to Jetson Nano which supports 4 USB 3.0 ports). However, I used the micro usb port of Jetson TX2 attaching a micro usb to USB 3.0 converter cable (provided along with Jetson TX2) and plugged my hub into that port where as I plugged my ZED camera to the regular single USB 3.0 port of Jetson TX2. The problem seems to be solved.

Myzhar, 2019-11-03:
@adiisawant have you solved the original problem with your ZED?
Can this issue be closed?

Myzhar, 2020-03-23:
Issue closed for inactivity

closed 2020-03-23 by Myzhar