stereolabs/zed-ros-wrapper #445: use of x,y,z coordinates of object detected using yolo in ros

virraaj, 2019-07-05:
I am trying to use zed camera with [darknet_ros](https://github.com/leggedrobotics/darknet_ros) package. I can successfully implement it without getting the x,y,z coordinates of objects. Can anyone give me a hint if I need those coordinates in this ros package, what can be my steps? Can I make use of any topic from *zed-ros-wrapper* to get them?

Myzhar, 2019-07-05:
Hi @virraaj,

you must write a new node that subscribes to the topics emitted by `darknet_ros` and to the ZED Depth image topic. 

You can begin from the "[zed_depth_sub_tutorial](https://www.stereolabs.com/docs/ros/depth_sensing/#depth-subscribing-in-c)" 

What is important for you is this part of the code of the tutorial:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/tutorials/zed_depth_sub_tutorial/src/zed_depth_sub_tutorial.cpp#L39-L40

You must replace `u` and `v` with the image coordinates of the point where you want to get the distance (the center of the bounding box?)

Let me know if you need further help

virraaj, 2019-07-08:
Hi @Myzhar , 
Thank you for your prompt reply. I tried using the center of the bounding box coordinates in *u* and *v*,  but I sometimes get the Index error like 
```
IndexError: index 962 is out of bounds for axis 0 with size 720
```
Also , can you suggest me if there is any way or starting steps to interface [zed-yolo](https://github.com/stereolabs/zed-yolo) package with this zed-ros-wrapper. That would be more interesting because I can get those 3d coordinates readily available from that package.

Myzhar, 2019-07-08:
About the wrong index maybe that there is something wrong in your code. If the bounding box is inside the image then the index of its center cannot be out of range.

About using the zed-yolo package, you must transform it in a ROS node, adding the code to publish the information you like as ROS topics.

virraaj, 2019-07-09:
Currently, I am using the following 
```
a = ((self.object_detected.bounding_boxes[0].xmax - self.object_detected.bounding_boxes[0].xmin)/2) + self.object_detected.bounding_boxes[0].xmin
b = ((self.object_detected.bounding_boxes[0].ymax - self.object_detected.bounding_boxes[0].ymin)/2) + self.object_detected.bounding_boxes[0].ymin
print('depth to person:{dist} m'.format(dist=self.depth_array[a,b]))
```
The *object_detected.bounding_boxes[0].x and y* are the bounding box coordinates from *darknet_ros*
I am getting *depth_array* from cv2bridge to get the distance in meters.  
Now, the problem is that the distance keeps varying over a wide range even if object is stationery. 
Can you show me what I am doing wrong here. 
Let me know if some more information is needed.

Myzhar, 2019-07-09:
```Now, the problem is that the distance keeps varying over a wide range even if object is stationery.```
Can you give me a few numbers to understand how wide is the range.

Furthermore you must check that the center of the bounding box has a valid depth. Not all the depth map points contain a valid measure.

virraaj, 2019-07-09:
It is varying between 2 meters to 10 meters.

Myzhar, 2019-07-09:
And what is the correct distance? Is the center of the bounding box effectively on the object?
It is a bounding box, so not all the points inside it belongs to the object

virraaj, 2019-07-09:
The correct distance is 0.8 m. If I am stationery, the range is 0.3 to 3 m. And on slight movement it increases to 10 m as well. Yes, center of bounding box seems to be effective on object. 
```
    xmin: 349
    ymin: 30
    xmax: 793
    ymax: 552
('Xc', 571)
('Yc', 287)
```

Myzhar, 2019-07-09:
I need images to really understand what's happening. It seems that there is something wrong with depth extracting

virraaj, 2019-07-09:
When it detects only one object, it gives accurate distances. I guess it would be reading coordinates of other objects before. I need to work around in it. But Yes, if I restrict the network to detect only one class of object, I get accurate distance. Thanks.

Myzhar, 2019-07-09:
Close the issue when it's solved

closed 2019-07-09 by virraaj

youssefmohana, 2021-05-13:
Iwork with darknet package in ros and zed 2 ineed help for make darknet work with zed to get distance from object x, y, z

PauloMendes33, 2021-12-23:
Your problem is because your bounding boxes sizes are out of the image size. Width and height may be out of image size. You need to set the limit of the image size to the bounding box.

Are you creating a map with the bounding boxes detected in ROS?